teemuatlut/TMCStepper #134: ESP32 with BTT TMC2208 v3.0: Abort window too short

holgerpieta, 2020-06-24:
After some hours of frowning and head scratching I managed to get my ESP32 to talk to the BTT TMC2208 v3.0.

First problem had nothing to do with the software: VIO needs to be 5 V. UART can have 3.3 V level, but nothing works when VIO is also just 3.3 V. Ah, well.

After I got that fixed I was able to use the Watterott Configurator and a generic USB-UART interface, so the TMC itself was fine.
But still no answer to the ESP32: Monitoring the UART using the USB interface showed that the TMC answered, but the software didn't like what it got and sometimes even talked over the answer with its second try.

Long story short: After some debugging I got it working by setting TMC2208Stepper::abort_window to 255. I haven't tried to find the lowest possible number, but 5 is definitely too short.

There is some discussion about the abort window in #24, so maybe there are people who know why abort_windows should be anything less than 255.

teemuatlut, 2020-06-24:
If you supply 5V to VIO then the Serial response set will also be at 5V. Not good if your MCU can't handle it.

The Serial based TMC drivers have a configurable time to wait before they start sending their reply.
By default this is 8 bit times and at a low 19200 baud this is around 400us or 0.4ms. Much MUCH less than 255 microseconds. Even that 5ms is ten times the expected delay.
I can try looking at the ESP compatibility next month when I have my vacation days.

holgerpieta, 2020-06-24:
Hm, ah, well, let's hope I'm not going to burn the ESP32 then. Maybe I have to think about a level shift or something. Or maybe measure first. Anyway, the drivers don't run at 3.3 V, so I don't have a choice.

But back to the real topic:
Your thinking about timing was exactly my thinking, so I tried a lot of other things before I added a lot of debug output to the _sendDatagram function and then I found the problem. I'll add some more speaking debug outputs later and put a log up here, then you can trace what happens.
The problem seemed to be that the receiver got a lot of 0xFF before the actually data and that caused the timeout to trigger before the end of the data was reached.
Anyway, don't do anything before I posted the logs, maybe I'm barking up the wrong tree.

holgerpieta, 2020-06-25:
OK, it's so many wrong trees, it's getting somewhat embarrassing.

The driver works just fine with 3.3 V on VIO, dunno why it didn't before.
But even when powered with 5 V, the voltage level on UART is still 3.3 V, so no danger for the ESP32, as it seems. Though I've only checked with a multi-meter, not with an oscilloscope.

After removing the library version installed by PlatformIO via the lib_debs mechanism and cloning this repository, communication worked just fine without any changes. So I compared versions and lib_debs wants to install the same 0.7.0. So I returned to the lib_debs version and voila, this also worked. As I've installed the old lib_debs version more then 4 days ago, it's probably still been version 0.6.2.
So, whatever was the problem, it was solved between 0.6.2 and 0.7.0.

So, sorry about the noise, thanks for the fix (whatever it was) and please just close this ticket.

teemuatlut, 2020-07-01:
Good to know you got it working.

As the original poster of the ticket, you too also have the possibility to close your own tickets.

closed 2020-07-01 by teemuatlut