teemuatlut/TMCStepper #135: Uncontrolled movements in the home and bed leveling v0.7.0

sc44, 2020-06-27:
Driver Bigtreetech TMC2209 v2.
Marlin 2.0.5.3 

After the update to 0.7.0, homing and bed leveling are no longer possible. The axes sometimes run the wrong way round and hit the stops at full speed. After downgrade to 0.6.2 everything is fine again.

teemuatlut, 2020-06-27:
If you know how to checkout specific commits, it would help me out a lot to know the specific one where the behavior changes.

teemuatlut, 2020-06-27:
If you could try this branch and let me know if it works better https://github.com/teemuatlut/TMCStepper/tree/splitPrePostCommunication

sc44, 2020-06-27:
Hello,
didn't have that much time for lunch today, had to go to work. I have tested this commit first.
https://github.com/teemuatlut/TMCStepper/tree/76c22be30d56b48b904de763fe70198dd0dae80b
but the problems come after this
https://github.com/teemuatlut/TMCStepper/tree/9e56b6e388d703a6c88b267575e5a492bf7201b9
re-reducing the problem is easy. Twice homing, first is ok the second time all axes move very quickly in the wrong direction. ABL after homing too. Mainboard is a Bigtreetech SKR V1.1 pro. Homing is sensorless except for Z. Hope this helps a little.

sc44, 2020-06-27:
I only saw that you changed the source. I have now tested the version https://github.com/teemuatlut/TMCStepper/tree/splitPrePostCommunication, looks good so far, homed several times and everything went normally. Thank you for the great job.

closed 2020-06-29 by teemuatlut