teemuatlut/TMCStepper #136: ESP lost connection after updating the TMC2130 Library.

Talha909, 2020-06-29:
I have installed ESP 3D with my SKR 1.4 Turbo with TMC2130. It was working good. Yesterday I installed the new bugfix. The stepper Driver also updated. I made the changes and upload the firmware again. Before  tmc2130 library 
[Marlin.zip](https://github.com/teemuatlut/TMCStepper/files/4844514/Marlin.zip)
undated my esp is working good. But after update my esp loss the connection the with board. I connected with the esp but there is no communication with esp and printer. I did not get the printer configuration from esp and no command is working. There diver of library of TMC2130 also updated

teemuatlut, 2020-06-29:
Try the 0.7.1 version or the master branch.

Talha909, 2020-06-30:
I try to updated the liberay in platefom.ino. but i cant see the 0.7.1. How i update the libraray

teemuatlut, 2020-07-01:
`.platformio` file has the dependencies listed. Version 0.7.1 got listed to PIO today.

Talha909, 2020-07-01:
TODAY I updated the liberay. But i still see the 0.70 when comile.

teemuatlut, 2020-07-01:
Try deleting the libdeps folder to force a re-download.

closed 2020-10-17 by Talha909