teemuatlut/TMCStepper #137: TMC2208 Marlin's TMC_ADV set

zonalimitatore, 2020-06-30:
Hi there.
I see that in marlin's (config_adv),i can set my params for my stepstick.
I would like set irun,ihold,toff,delay,autograd,hend,hstrt,tbl
I don't know what parameters are supported,and the sintax for compile it right.
Thanks

teemuatlut, 2020-06-30:
https://teemuatlut.github.io/TMCStepper/class_t_m_c2208_stepper.html

irun and ihold will be overriden by eeprom settings. Either by the defaults or the values from memory.

toff, hend and hstrt are best set with the chopper parameters setting, unless you need to use different settings for drivers.

The syntax is a multiline define
```cpp
   #define TMC_ADV() { \
     stepperX.diag0_otpw(1); \
     stepperY.intpol(0); \
  }
```

zonalimitatore, 2020-06-30:
Thanks.
I try to use stepperX.pwm_autograd(true) or stepperX.pwm_aurograd(1), but have error when compile

teemuatlut, 2020-06-30:
Remember to include the backslashes as it's a multiline define.

zonalimitatore, 2020-06-30:
Instead set in marlin,can i set value in trinamic.cpp under my stepper type?

zonalimitatore, 2020-07-01:
Ok,i've set it into marlin.
I got errors because,for some way, i've uncomment "SOFTWARE_DRIVER_ENABLE" 
Remain errors like "has_rx" line 162 of tmc_util.h,and line 824 of TMCStepper.h,but driver work
Thank you

closed 2020-07-01 by teemuatlut