teemuatlut/TMCStepper #139: TMC2209 StallGuard Example for ESP32

johnsmakar, 2020-07-07:
Can you create a example sketch for using the StallGuard feature on a 2209 for an ESP32? I’m not sure how the interruption registers for the mega translate to the esp. Or if you know of a good resource where I can learn to do it on my own.

edwardocano, 2020-07-09:
Hello!To  Work with interruption in ESP32 is a bit different. this is a Example with stallguard. The function giro_normal works like in atmega328p. Tomorrow i will upload this example to my repository, this is just a draft version. 

`#include <Arduino.h> 
#include <TMCStepper.h>
#include <HardwareSerial.h>
//Cambiar hardwareserial.cpp pin definition
#define EN_PIN           14
#define DIR_PIN          26 // Direction para el ESP
#define STEP_PIN         27 //Step para el ESP
//#define SERIAL_PORT      Serial2
#define DRIVER_ADDRESS 0b00
#define R_SENSE        0.11f 
#define STALL_VALUE       5

hw_timer_t * timer1 = NULL; //se declara un apuntador de tipo para la configuracion del timer 
HardwareSerial motor1(1);    //Si le pongo HardwareSerial Serial1(1); me marca error ya que s_1 ya existe. 
HardwareSerial motor2(2);
TMC2209Stepper driver(&motor2, R_SENSE, DRIVER_ADDRESS);

//void giro_normal();

void IRAM_ATTR onTimer() {
  //digitalWrite(STEP_PIN, !digitalRead(STEP_PIN));
}
 
void setup() {
  
 
  Serial.begin(115200);
  while(!Serial); 

  motor2.begin(115200);
  //driver.beginSerial(115200); //use in case software serial.

  pinMode(STEP_PIN, OUTPUT);
  pinMode(DIR_PIN, OUTPUT);
  pinMode(EN_PIN , OUTPUT);

  digitalWrite(EN_PIN,LOW);
  
  driver.begin();
  driver.toff(4);
  driver.blank_time(24);
  driver.rms_current(400); // mA
  driver.microsteps(16);
  driver.TCOOLTHRS(0xFFFFF); // 20bit max
  driver.semin(5);
  driver.semax(2);
  driver.sedn(0b01);
  driver.SGTHRS(STALL_VALUE);
  

  giro_normal();
}
 
void loop() {
  
  static uint32_t last_time=0;
  uint32_t ms = millis();

  if((ms-last_time) > 100) { //run every 0.1s
    
    last_time = ms;

    Serial.print("0 ");
    Serial.print(driver.SG_RESULT(), DEC);
    Serial.print(" ");
    Serial.println(driver.cs2rms(driver.cs_actual()), DEC);
  }

}

void giro_normal(){
  cli();
  timer1 = timerBegin(3, 8,true); // Se configura el timer, en este caso uso el timer 4 de los 4 disponibles en el ESP(0,1,2,3)
                                 // el prescaler es 8, y true es una bandera que indica si la interrupcion se realiza en borde o en nivel
  timerAttachInterrupt(timer1, &onTimer, true); //Se vincula el timer con la funcion AttachInterrup 
                                               //la cual se ejecuta cuando se genera la interrupcion
  timerAlarmWrite(timer1, 10000, true); //En esta funcion se define el valor del contador en el cual se genera la interrupción del timer
  timerAlarmEnable(timer1); //Función para habilitar el temporizador.
  sei();
}`

johnsmakar, 2020-07-09:
@edwardocano okay, awesome. Thanks so much...

when compiling, do you still get a warning that the avr architecture may not be comparable with esp32 architecture? I'm using the most up to date TMCStepper.h (0.7.1)

edwardocano, 2020-07-09:
Yes, but i have tested almost all the functions of tmc2209 and i have not found any problem.

johnsmakar, 2020-07-11:
@edwardocano when you get a chance, could you upload the example :)

edwardocano, 2020-07-11:
Hello! i have added this repository with a usefull example. I recommend you to use the uart 2 , uart 1 have some issues due to share some pins with flash memory. [example]
https://github.com/edwardocano/Esp32-TMC2209

johnsmakar, 2020-07-11:
On line 34, you write the Enable pin to HIGH, which is Deactivate.

johnsmakar, 2020-07-11:
@edwardocano On line 34, you write the Enable pin to HIGH, which is Deactivate.

johnsmakar, 2020-07-11:
@edwardocano I am not sure if I am doing something incorrect, 
these are my pins

`#define EN_PIN_2           32          // Enable
#define DIR_PIN_2          12          // Direction
#define STEP_PIN_2         14          // Step`

the stepper doesn't stall even with the stall value all the way down.

edwardocano, 2020-07-11:
I do not understand, the motor keeps moving when you disable the enable pin?

johnsmakar, 2020-07-11:
@edwardocano 
When EN_PIN is HIGH the motor is NOT active

When the EN_PIN is LOW, the motor is active...

When the motor is active, and I test the sketch, the motor doesn't stall

All I see in the Serial monitor is 0 0 55

What am I missing?

edwardocano, 2020-07-11:
Yes, when the enable pin is low the motor must be active. Try this, add this line to your code and tell me what value gets result_1
`uint8_t result_1 = driver2.test_connection();`

johnsmakar, 2020-07-11:
@edwardocano I get 2

edwardocano, 2020-07-11:
You have connected something wrong, do you connect the PDN pin in the correct way? What is the voltage of the Vio pin of stepper driver.

johnsmakar, 2020-07-11:
@edwardocano Ah, I think that's my problem haha...

I am connecting my EN, DIR, and STEP pins, I didn't know to use the PDN pin.

Which are the correct pins I should be using?

edwardocano, 2020-07-11:
I would say all, but i do not know what do you want to do with the drivers.

johnsmakar, 2020-07-11:
@edwardocano I just want to be able to set Speed and Position for a linear rail.

But for the example sketch for stallGuard, where should DIAG_PIN_2 , EN_PIN_2,DIR_PIN_2, and STEP_PIN_2 be connected to so I can get it to work Properly?

edwardocano, 2020-07-11:
Read the datasheet, diag,en,dir and step pin must be connected to the driver pins with same name. And the comunication between the esp and the driver is through uart, but driver has something named PDN in the datasheet explains how to connect PDN to uart.

johnsmakar, 2020-07-11:
@edwardocano Sorry for all the questions, I'm new to this... After looking at the datasheet

My PDN pins are wired to pins 16, 17 on the esp which correspond to uart 2 rx/tx and VIO is getting 5v... 

Does that sound right? I don't want to accidentally mess something up.

edwardocano, 2020-07-11:
The vio of the driver must be connected to 3.3, you are communicating esp32 and driver, and the eap32 just support 3,3 voltage level

johnsmakar, 2020-07-11:
@edwardocano just changed vio to 3.3v, I don't see a change in anything. What should I be looking out for?

edwardocano, 2020-07-11:
If you get the value of 0 when  you send the test code the comunication is enabled, now you can read the data of diag pin.

johnsmakar, 2020-07-12:
@edwardocano DIAG_PIN_2 is an input, how do I read it?

closed 2020-07-12 by johnsmakar