teemuatlut/TMCStepper #140: Unable to get motor to spin

changedsoul, 2020-07-10:
I am trying to use this "Software_SPI" example, but i have tried others as well I found on the net and absolutely nothing I do will make my motor spin.
Can someone please help me figure out why. This is driving me NUTS!.
I thought it would be as simple as connecting the wires up and loading the code into the arduino, but nothing works.
I have tried with TMC2130's and TMC5160's and nothing happens.

I can drop them into my printers ramps board and they work, so I know the drivers are not faulty.
I am not sure what other info to give.

riversedge, 2020-07-29:
Same issue.   I eventually ended up using Hardware SPI by connecting it to ports on a Mega2560 SDI -> 51 (MOSI), SDO -> 50 (MISO), SCK -> 52.   Even hooked up like this and trying to use the software SPI initialization didn't work.   I think there might be an issue with the SW_SPI class and how it communicates.   You could try a minor rewrite using something like SoftSPI from here: https://github.com/MajenkoLibraries/SoftSPI/blob/master/src/SoftSPI.cpp

I didn't bother debugging any more as IMO hardware SPI is preferable anyway and it supports multiple devices via the chip select.  But FWIW I can also replicate your issue.

changedsoul, 2020-10-26:
Anyone else get this to work? I still cannot for life of me get the TMC5160 to work. 
I was able to get the 2130 to spin, but the 5160 just will not work. 
I am trying now the “Live Tune” example and just will not power up. Please can anyone help shed light on this?

riversedge, 2020-10-26:
I was able to get it to work fairly well, but I will say that the stallguard didn’t really work as I had hoped.   That was by far the most unreliable part and I will also say that some of the chips I got were better than others.   

Are you just trying to get it to control the motor, or are you trying to use stallguard for some application?

changedsoul, 2020-10-26:
At the moment I am trying to understand the chips. 
I am using in my 3D printer and are working fine. My goal was to control a motor with one using the arduino to start learning how to configure them and understand them better. 
My problem is all the code examples fail to get the motor to even move with spi (hardware or software)
I wanted to use the live tune example because it has example of you making changes and updating the chip. I feel this might be good to start learning how to read and write the chip registers etc, but if I cannot even get it to move, I don’t even know where to begin to look for the problem. 

Like it said, if I comment out tmc2130 and swap out the chips, the code moves the motor connected to a tmc2130, but if I try and configure it for the tmc5160, I get nothing from it. 

So confused.

riversedge, 2020-10-26:
This should get you started, though you will need to adjust the current levels for your motor along with any other wire changes:

```
#include <TMCStepper.h>

#define R_SENSE            0.075f  // Driver matched
#define STALL_VALUE            12  // [-64..63]
#define TARGET_PEAK_CURRENT  2800
#define TARGET_RMS_CURRENT   2000  // (TARGET_PEAK_CURRENT * 0.707)
#define STALL_PIN   D32            // Stall pin (Diag1)

TMC5160Stepper stepperDriver(ONBOARD_CS_PIN, R_SENSE); // Hardware SPI (Mega2560 SDO=50, SDI=51, SCK=52)

bool TMC5160ControllerSetup()
{
   pinMode(ONBOARD_ENABLE_PIN, OUTPUT);
   pinMode(ONBOARD_STEP_PIN, OUTPUT);
   pinMode(ONBOARD_CS_PIN, OUTPUT); 
   pinMode(ONBOARD_DIR_PIN, OUTPUT);
   pinMode(MISO, INPUT_PULLUP);
  
   // Set chip select pin
   pinMode(ONBOARD_CS_PIN, OUTPUT);
   digitalWrite(ONBOARD_CS_PIN, HIGH);

   #ifdef USE_STALLGUARD2
   pinMode(STALL_PIN, INPUT_PULLUP);
   #endif

   SPI.begin();  
   stepperDriver.begin();

   stepperDriver.reset();
   stepperDriver.toff(0);
   stepperDriver.push();
   digitalWrite(ONBOARD_ENABLE_PIN, LOW);
   stepperDriver.toff(3);

   CHOPCONF_t chopconf{0};
   chopconf.tbl = 2;
   chopconf.toff = 3;
   chopconf.intpol = true;
   chopconf.hend = 2;
   chopconf.hstrt = 0;
   chopconf.dedge = false;
   stepperDriver.CHOPCONF(chopconf.sr);

   stepperDriver.iholddelay(10);// Current during hold = full
   stepperDriver.irun(31);    // Current during run = full
   stepperDriver.TPOWERDOWN(128); // ~2s until driver lowers to hold current
   stepperDriver.chm(0);    // Chopper mode (0 = standard, 1 = constant off time w/ fast decay time)
   stepperDriver.hstrt(4);  // Chopper Hysteresis low end
   stepperDriver.hend(2);   // Chopper Hysteresis high end
   stepperDriver.tbl(2);    // Chopper configuration (Set comparator blank time to 16, 24, 36 or 54 clocks) (1 or 2 value for typical - 1=24 tclk)
  
   stepperDriver.blank_time(24);
   stepperDriver.rms_current(TARGET_RMS_CURRENT); // mA
   stepperDriver.microsteps(0);

   stepperDriver.en_pwm_mode(true);      // Toggle stealthChop on TMC2130/2160/5130/5160
   stepperDriver.TPWMTHRS(500);   
   stepperDriver.TCOOLTHRS(0xFFFFF); // 20bit max - Lower velocity threshold for switching on CoolStep and stop on stall. Below this velocity CoolStep becomes disabled (not used in STEP/DIR mode). Adapt to the lower limit of the velocity range where StallGuard2 gives a stable result.
   stepperDriver.THIGH(0);    // Upper velocity threshold value for CoolStep and stop on stall. Above this velocity CoolStep becomes disabled. Adapt to the velocity range where StallGuard2 gives a stable result.
   stepperDriver.semin(5);    // [0-15] If the StallGuard2 result falls below SEMIN*32, the motor current becomes increased to reduce motor load angle. 0 = smart current control CoolStep off
   stepperDriver.semax(2);    // [0-15] If the StallGuard2 result is equal to or above (SEMIN+SEMAX+1)*32, the motor current becomes decreased to save energy.
   stepperDriver.sedn(0b01);  // For each 8 StallGuard2 values decrease by one

   TMC2160_n::PWMCONF_t pwmconf{0};
   pwmconf.pwm_lim = 12;
   pwmconf.pwm_reg = 8;
   pwmconf.pwm_autograd = true;
   pwmconf.pwm_autoscale = true;
   pwmconf.pwm_freq = 0b01;
   pwmconf.pwm_grad = 14;
   pwmconf.pwm_ofs = 36;
   stepperDriver.PWMCONF(pwmconf.sr);
   stepperDriver.drvstrength(0b11);   // 00: weak  01: weak+TC (medium above OTPW level)  10: medium  11: strong  

   #ifdef USE_STALLGUARD2
   stepperDriver.diag1_stall(true);  // Enable diag1 on stall
   stepperDriver.freewheel(false);
   #else
   stepperDriver.sfilt(true);
   #endif

   stepperDriver.GSTAT(); // Clear GSTAT
   stepperDriver.RAMP_STAT();

   int stepperInitResult = stepperDriver.test_connection();

   if (stepperInitResult != 0) return false;

   return true;
}
```

Call that first and it will init the 5160 stepper driver.   Then I suggest using another library like AccelStepper inited with (AccelStepper::DRIVER, STEP_PIN, DIR_PIN) for movement.  I again would HIGHLY recommend Hardware SPI for reliability.  I can confirm this works in a circuit and the TMCStepper code itself also is good (though I'm not using stallguard at the moment).   If you aren't wired with SPI correctly it won't set up the driver properly and things won't run.   If the power isn't set up on both the motor and chip side then it also won't initialize correctly.  In the above example you can look at stepperInitResult's value and it'll tell you why it failed.

changedsoul, 2020-10-26:
Thanks a lot for talking the time, I’ll give a go tomorrow. I think I have everything wired up correctly. It was working for the 2130 chip. The pin outs are the same from what it looks like. I assumed simply commenting the 2130 and uncommenting the 5160 should have worked. 

If this works for me it will be a big step forward in at least knowing it’s working and giving me a baseline to start building on. 

Thanks again. 
I’ll post back my results.

changedsoul, 2020-10-26:
Well I am still failing to enable these drivers. I dont know what else might be wrong. I can verify the driver works in my printer, so I know its not fried, but when I connect to the arduino I get nothing.
This is the exact code I uploaded to the arduino. I am using The Arduino Duemilanove.

>  In the above example you can look at stepperInitResult's value and it'll tell you why it failed.
When I look at the status, I just get "1".
```
#include <TMCStepper.h>

#define R_SENSE            0.075f  // Driver matched
#define STALL_VALUE            12  // [-64..63]
#define TARGET_PEAK_CURRENT  2800
#define TARGET_RMS_CURRENT   2000  // (TARGET_PEAK_CURRENT * 0.707)
//#define STALL_PIN   D32            // Stall pin (Diag1)

#define ONBOARD_ENABLE_PIN       2 // Enable
#define ONBOARD_DIR_PIN          3 // Direction
#define ONBOARD_STEP_PIN         4 // Step
#define ONBOARD_CS_PIN           10 // Chip select
//#define SW_MOSI          11 // Software Master Out Slave In (MOSI)
//#define SW_MISO          12 // Software Master In Slave Out (MISO)
//#define SW_SCK           13 // Software Slave Clock (SCK)

TMC5160Stepper stepperDriver(ONBOARD_CS_PIN, R_SENSE); // Hardware SPI (Mega2560 SDO=50, SDI=51, SCK=52)

bool TMC5160ControllerSetup()
{
   pinMode(ONBOARD_ENABLE_PIN, OUTPUT);
   pinMode(ONBOARD_STEP_PIN, OUTPUT);
   pinMode(ONBOARD_CS_PIN, OUTPUT); 
   pinMode(ONBOARD_DIR_PIN, OUTPUT);
   pinMode(MISO, INPUT_PULLUP);
  
   // Set chip select pin
   pinMode(ONBOARD_CS_PIN, OUTPUT);
   digitalWrite(ONBOARD_CS_PIN, HIGH);

   #ifdef USE_STALLGUARD2
   pinMode(STALL_PIN, INPUT_PULLUP);
   #endif

   SPI.begin();  
   stepperDriver.begin();

   stepperDriver.reset();
   stepperDriver.toff(0);
   stepperDriver.push();
   digitalWrite(ONBOARD_ENABLE_PIN, LOW);
   stepperDriver.toff(3);

   CHOPCONF_t chopconf{0};
   chopconf.tbl = 2;
   chopconf.toff = 3;
   chopconf.intpol = true;
   chopconf.hend = 2;
   chopconf.hstrt = 0;
   chopconf.dedge = false;
   stepperDriver.CHOPCONF(chopconf.sr);

   stepperDriver.iholddelay(10);// Current during hold = full
   stepperDriver.irun(31);    // Current during run = full
   stepperDriver.TPOWERDOWN(128); // ~2s until driver lowers to hold current
   stepperDriver.chm(0);    // Chopper mode (0 = standard, 1 = constant off time w/ fast decay time)
   stepperDriver.hstrt(4);  // Chopper Hysteresis low end
   stepperDriver.hend(2);   // Chopper Hysteresis high end
   stepperDriver.tbl(2);    // Chopper configuration (Set comparator blank time to 16, 24, 36 or 54 clocks) (1 or 2 value for typical - 1=24 tclk)
  
   stepperDriver.blank_time(24);
   stepperDriver.rms_current(TARGET_RMS_CURRENT); // mA
   stepperDriver.microsteps(0);

   stepperDriver.en_pwm_mode(true);      // Toggle stealthChop on TMC2130/2160/5130/5160
   stepperDriver.TPWMTHRS(500);   
   stepperDriver.TCOOLTHRS(0xFFFFF); // 20bit max - Lower velocity threshold for switching on CoolStep and stop on stall. Below this velocity CoolStep becomes disabled (not used in STEP/DIR mode). Adapt to the lower limit of the velocity range where StallGuard2 gives a stable result.
   stepperDriver.THIGH(0);    // Upper velocity threshold value for CoolStep and stop on stall. Above this velocity CoolStep becomes disabled. Adapt to the velocity range where StallGuard2 gives a stable result.
   stepperDriver.semin(5);    // [0-15] If the StallGuard2 result falls below SEMIN*32, the motor current becomes increased to reduce motor load angle. 0 = smart current control CoolStep off
   stepperDriver.semax(2);    // [0-15] If the StallGuard2 result is equal to or above (SEMIN+SEMAX+1)*32, the motor current becomes decreased to save energy.
   stepperDriver.sedn(0b01);  // For each 8 StallGuard2 values decrease by one

   TMC2160_n::PWMCONF_t pwmconf{0};
   pwmconf.pwm_lim = 12;
   pwmconf.pwm_reg = 8;
   pwmconf.pwm_autograd = true;
   pwmconf.pwm_autoscale = true;
   pwmconf.pwm_freq = 0b01;
   pwmconf.pwm_grad = 14;
   pwmconf.pwm_ofs = 36;
   stepperDriver.PWMCONF(pwmconf.sr);
   stepperDriver.drvstrength(0b11);   // 00: weak  01: weak+TC (medium above OTPW level)  10: medium  11: strong  

   #ifdef USE_STALLGUARD2
   stepperDriver.diag1_stall(true);  // Enable diag1 on stall
   stepperDriver.freewheel(false);
   #else
   stepperDriver.sfilt(true);
   #endif

   stepperDriver.GSTAT(); // Clear GSTAT
   stepperDriver.RAMP_STAT();

   int stepperInitResult = stepperDriver.test_connection();

   if (stepperInitResult != 0) return false;

   return true;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(500000);
  if (TMC5160ControllerSetup()) {
    Serial.println("Motor returned True");
  }
  else {
    Serial.println("Motor failed to initialze");
    Serial.println(stepperDriver.test_connection());
  }
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(ONBOARD_STEP_PIN, HIGH);
  delay(10);
  digitalWrite(ONBOARD_STEP_PIN, LOW);
  delay(10);
}
```

changedsoul, 2020-10-30:
Hey so do you by chance know how to check these with an oscilloscope?
I’ve been trying to do the hysteresis start and end using the scope like they show in the data sheet, but I just cannot for the life of me get a wave form to be produced?

teemuatlut, 2020-11-04:
First make sure you have working communication. Start with a very basic program that doesn't do anything besides print you the driver firmware version (`.version()`).
I made a comment on another thread about controlling the TMC5160 and you can copy the connection detection from there: https://github.com/teemuatlut/TMCStepper/issues/157#issuecomment-721960004 (With minor changes depending on which branch you're using).