teemuatlut/TMCStepper #141: Stepper "dies" after running for a few, have to completely stop external power and reset to get it to run

riversedge, 2020-07-29:
I'm having a problem where seemingly randomly the motor will basically die out.   The RMS voltage slowly climbs but the motor won't run.   To get it to run again I have to power off both the board and the motor power supply (just resetting the board doesn't work).  Either there's something wrong with the TMCStepper (0.7.1) or something is just horribly unreliable about the chipset (which doesn't make sense to me because it seems to be used frequently).

I've attached the slightly modified program from the example that's:
1) Been SGT tuned to the motor
2) Wired up to hardware SPI
3) Added some code to try and activate parts of StallGuard on the TMC5160.

Whatever makes it stop does not seem to be StallGuard, nor the temperature.   I'm at a loss after several days of troubleshooting as to what's going on and why a board reset doesn't seem to have any effect.

Output from the program:
```
20:20:34.707 -> 
20:20:34.707 -> Start...
20:20:34.707 -> Driver StallGuard Stop: 0
20:20:34.707 -> Target RMS current: 1979.6000976562
20:20:34.707 -> 
20:20:34.707 -> Testing connection...OK
20:20:34.817 -> 101 00011110000000 10000000000011110000000001111011 119 980 0 1
20:20:34.931 -> 101 00011110000000 10000000000011110000000001110111 118 980 0 1
20:20:35.003 -> 101 00011110000000 10000000000011110000000001111110 122 980 0 1
20:20:35.108 -> 101 00011110000000 10000000000011110000000001110010 120 980 0 1
20:20:35.216 -> 101 00011110000000 10000000000011110000000001111101 117 980 0 1
20:20:35.317 -> 101 00011110000000 10000000000011110000000001110010 122 980 0 1
20:20:35.418 -> 101 00011110000000 10000000000011110000000001111001 122 980 0 1
20:20:35.528 -> 101 00011110000000 10000000000011110000000001111111 120 980 0 1
20:20:35.628 -> 101 00011110000000 10000000000011110000000001110110 118 980 0 1
20:20:35.736 -> 101 00011110000001 00000000000100000000001010011011 637 1046 0 1
20:20:35.811 -> 101 00011110000001 00000000000011110000001010100001 652 980 0 1
20:20:35.918 -> 101 00011110000000 00000000000011110000001010000111 664 980 0 1
... Many lines like this - same register flags, etc...
20:20:50.707 -> 101 00011110000001 00000000000011110000001010001000 670 980 0 1
20:20:50.774 -> 101 00011110000001 01100000000111110000000000000000 0 1970 0 1
20:20:50.885 -> 101 00011110000000 01100000000111110000000000000000 0 1970 0 1
20:20:50.996 -> 101 00011110000001 01100000000111110000000000000000 0 1970 0 1
... Never runs the motor again.
```

The lines are the registers (in binary): 
```
GSTAT: 101
RAMP_STAT: 00011110000001
DRV_STATUS:  01100000000111110000000000000000
```

I looked at the spec sheet and can't find anything that obviously points to an issue.   The driver status just seems to indicate the informational message it's open (because the motor is stopped) and that current multiplier is cranked all the way.

[StallGuard-TMC5160.txt](https://github.com/teemuatlut/TMCStepper/files/4991760/StallGuard-TMC5160.txt)

teemuatlut, 2020-07-29:
I'll see if I can test the TMC5160 this week myself. At a quick glance I don't see anything wrong with your code.

Resetting just the MCU doesn't work because the driver has memory itself and that does not get reset. You can reset the fault conditions by disabling the driver with toff=0 or with the enable pin.
Your CS values are pinned because the library will primarily try to set the current with global scaler and will only modify the irun value if there's not enough range.
Open load flags don't disable the coils, only shorts to ground or the supply voltage can do that. So we'll have to see what the reason could be why the motor wouldn't run anymore.

How long does it take for the motor to stop?

riversedge, 2020-07-29:
When it stops seems random.   In the above example it stopped after about 13s.   Ran it again just now and it went for about 33s from cold start (everything off for hours).  I've seen it go shorter (almost immediate) and longer (a few minutes).  I have tried the enable pin (off to back on) and modified the "0/1" serial receive code to also call driver.toff(0) on "0" and then re-setup things on "1" and that did not get the motor moving again on it's own.  Only if I power off the external motor voltage then hit "1" with the below code will it start for a few more seconds before stopping.

i.e., Tested it with:
```
      if (read_byte == '0')      { TIMSK1 &= ~(1 << OCIE1A); stopped = true; driver.toff(0); /*digitalWrite( EN_PIN, HIGH );*/ stalled = false;}
      else if (read_byte == '1') { TIMSK1 |=  (1 << OCIE1A); stopped = false; driver.toff(5);/*digitalWrite( EN_PIN,  LOW );*/ stalled = false;}
```
and also
```
      if (read_byte == '0')      { TIMSK1 &= ~(1 << OCIE1A); stopped = true; driver.toff(0); /*digitalWrite( EN_PIN, HIGH );*/ stalled = false;}
      else if (read_byte == '1') { TIMSK1 |=  (1 << OCIE1A); stopped = false; SPI.end(); setup();/*digitalWrite( EN_PIN,  LOW );*/ stalled = false;}
```

FWIW this is also on a stable (no vibration) platform.   Worried about the possibility of a random short I went from a breadboard mounted chip to soldering it onto a protoshield with no difference in behavior.

teemuatlut, 2020-07-29:
I used your code with only a few changes into the pin definitions and the motor has been spinning for 25 minutes now. The stallGuard measurements seem to work as well.
The platform was a Mega2560, RAMPS 1.4 and a pre-production TMC5160 stepstick. The motor was a 2.8A NEMA23 [57HS56](https://www.vallder.com/en/?option=com_virtuemart&view=productdetails&virtuemart_category_id=38&virtuemart_product_id=385&Itemid=154).

Make sure to ground the TMC5160 CLK pin (clock, **not** SPI SCK) or try with another stepper motor.

riversedge, 2020-07-29:
Thanks!  I did ground the CLK pin and it doesn't seem to make a difference.   I'm using Mega2560 with the TMC5160 (BIGTREETECH DIRECT TMC5160 V1.2 SPI) directly wired to the Mega and a Nema17 17HS16 motor.   I'll try a different motor, but curious if there is anything running it through the RAMPS board might be doing around buffering or other such functionality?   It should work directly connected, right?

riversedge, 2020-07-30:
I tried a different motor, same problem.   Not sure if you have any other ideas or not.  Perhaps I'll try and find another driver - just very odd unless the RAMPS in the middle is doing something.

riversedge, 2020-08-01:
Can you tell me the brand of the TMC5160 stepstick you had no problems with?   Starting to think maybe this BigTree one is the issue - tried another and it produced similar results.   Hooked everything up to a RAMPS 1.4 board and it also didn't help.   Something very odd going on here and trying to get to the root of it.   Appreciate any help.

teemuatlut, 2020-08-01:
Most of my drivers are provided directly by Trinamic but I would always recommend to buy from Watterott. Though I do know that at least the BTT drivers use genuine ICs.

closed 2020-08-01 by riversedge