teemuatlut/TMCStepper #142: Use different hardware spi

mr-miky, 2020-08-01:
On the board that I designed and uses stm32F4 there are 3 spi hardware.
The TMC2130 that I use are connected to a dedicated spi hardware.
The TMCStepper library relies on the software spi or the Arduino SPI.
What I ask is if it is possible to implement a method to pass the pointer to a SPIClass other than the standard Arduino one.

I currently use a `SPIClass * TMC_HW_SPI` pointer declared extern and initialized in MarlinCore.cpp.
The problem is that I have to patch the TMCStepper library every time a new version is released.

If I want to use the Arduino SPI:

`TMC_HW_SPI = & SPI;`

if not :

`TMC_HW_SPI = new SPIClass (SPI1, TMC_SW_MOSI, TMC_SW_MISO, TMC_SW_SCK);`

If there was a method to pass the pointer it would be more portable without changing the library code.

teemuatlut, 2020-08-01:
See the upcoming Release_v1 branch. It's still very much alpha stage at this point though.

rohan-patel367, 2024-03-27:
Hello, has this been implemented? I am using a teensy 4.1 and would like to utilise SPI1 instead of SPI with a TMC5160. Just wondering if this is possible...

hayden-servaenergy, 2026-01-29:
Bumping this. I am also using a Teensy 4.1 and would like to use hardware SPI1 with my stepper driver, since the SPI 0 pins are being used for something else.