teemuatlut/TMCStepper #143: DRV_STATUS incorrect for TMC5160

riversedge, 2020-08-02:
0x6F for DRV_STATUS in datasheet shows 3 flags missing from the inherited TMC2130 struct.   For TMC5160 should have the following that includes s2vsa, s2vsb, stealth instead of the reserved 5 bits.

```
namespace TMC5160_n {
  struct DRV_STATUS_t {
    constexpr static uint8_t address = 0x6F;
    union {
      uint32_t sr;
      struct {
        uint16_t sg_result : 10;
        uint8_t            : 2;
        bool s2vsa : 1;
        bool s2vsb : 1;
        bool stealth : 1;
        bool fsactive : 1;
        uint8_t cs_actual : 5,
                          : 3;
        bool  stallGuard : 1,
              ot : 1,
              otpw : 1,
              s2ga : 1,
              s2gb : 1,
              ola : 1,
              olb : 1,
              stst : 1;
      };
    };
  };
}
```

teemuatlut, 2020-08-04:
Yup, you're absolutely correct.
This is fixed in the Release_v1 branch so it'll be included when the time is right.
https://github.com/teemuatlut/TMCStepper/blob/Release_v1/src/source/interfaces/TMC5160.hpp

johnsmakar, 2020-08-07:
Slight tangent, how you you wire the 5160/5161 for UART Mode? I can't seem to find the documentation for wiring diagram when its in UART mode. Thanks in advance!

riversedge, 2020-08-07:
Caveat- I’m not the expert but I have spent some quality time with all the datasheets.   The 5161 isn’t really out yet.  The 5160 data sheet for the chip has a single wire UART but I’ve not seen any of the board designs or stepper sticks have that support.  So I do not think you can on the 5160 - it’s designed to be SPI only when you look at the comparison charts.  5161 is supposed to add back in some ability to do config pins.

johnsmakar, 2020-08-07:
The only site I found that mentions uart mode is this one site m, they talk about which pins to jump to switch from SPI mode to UART...

https://wiki.fysetc.com/S5161_V1.1/

But there is no mention of which physical pin it is. The BigTreeTech drivers don’t have options for anything to be jumped.

riversedge, 2020-08-07:
If you look at the table you’ll see:

Configuration | CFG Pins or SP

And in the table below that none of them show enabling UART.

johnsmakar, 2020-08-07:
If you go to the ‘Wiring Diagram’ and you see the table it says if you set SPI = 0, SD = 0 that is for UART Mode.

riversedge, 2020-08-07:
On the chipset but not the stepper stick as far as I can see.  From what I’ve seen It’s not possible on the stepper sticks, but since I don’t have one I’m at the limit of being able to help you any further.   I looked the other day for my project and couldn’t see where the 5161 was even being sold broadly yet.

Possibly @teemuatlut has received an early release or has more useful information; however you’re asking for hardware support and this site is just around the Arduino driver library.  If you think it should be possible, might be best to submit something to FYSTEC who would be the experts in the hardware you’re looking at.