teemuatlut/TMCStepper #144: TMC 5160 / GRBL_ESP32 cannot increase current over .500

pcdonovan, 2020-08-13:


teemuatlut, 2020-08-13:
@bdring would likely know more about how the firmware interfaces with the library. I'd need to be more familiar with the code base to be able to say much.

On a generic level, the motor shouldn't stay disabled due to the driver overheating. This is what the datasheet says:
> After triggering the overtemperature sensor (ot flag), the driver remains switched off until the system temperature falls below the pre-warning level (otpw) to avoid continuous heating to the shutdown level.

However, a short detection will disable the driver until it is disabled and re-enabled.
> Once a short condition is safely detected, the corresponding driver bridge (A or B) becomes switched
off, and the s2ga or s2gb flag, respectively s2vsa or s2vsb becomes set.
To restart the motor, disable and re-enable the driver.

The diagnostic features of the TMC5160 are described in chapter 11 and can potentially help you debugging.

pcdonovan, 2020-08-13:
OK thank you, I guess I need to find out if bdring has programmed in TMC Debugging in the firmware.

pcdonovan, 2020-08-20:
@teemuatlut bdring did some testing with with his 5160's v1.1, he confirmed the ability to change the current without issue. My drivers are v1.2. I believe on the newer V1.2 the clk pin is pulled to ground, do you think that could  make a difference? I have another set of 5160's on their way tomorrow, Im hoping it will fix the issue but I am doubting it. I may try the 5161's if not I'm going to ditch the step sticks and go external drivers.

bdring, 2020-08-20:
My Trinamic controllers have that pin pulled to ground. You could check it.

I don't think the 5160 would work at all without it.

pcdonovan, 2020-08-20:
Bart, FYI I bent the clk pin outward and rinstalled the driver, and tested. They operate exactly the same.  Good thing is I have the new 5160's waiting for me when I get home from work. I'll update you when I pop them in.

pcdonovan, 2020-08-28:
@teemuatlut 
Upon further testing @bdring  could not duplicate the problem. I sent him a sent of drivers I was having issues with and he could not replicate the problem. I was reading the TMC5160 Datasheet and noticed there are some registers for Short_Conf, I had an issue with noise in the limit switch lines I'm wondering if its detecting a false short. Can these registers be changed in the tmcstepper library on my local machine when I compile the software. What directory would I be looking in.
[image]

pcdonovan, 2020-08-28:
I also found this and was thinking could it be the TOFF HEND and HSTRT setting?https://github.com/KevinOConnor/klipper/issues/2516

pcdonovan, 2020-08-28:
Any help would be appreciated

teemuatlut, 2020-08-31:
The library does support those registers, but I don't know how the ESP GRBL would conveniently configure the values. In Marlin you'd use the TMC_ADV macro.
Unfortunately I don't think your unique issues can be solved with software so my suggestion is to simplify your setup down to the very minimal. Use just the board and the drivers. Try with a different motor.

pcdonovan, 2020-09-01:
How can I set this to 0 to disable  bool diss2vs : 1; // TMC5160 only whenever I set to 0 It fails to compile

bdring, 2020-09-02:
This post has some interesting similarities to the problem.

roberthuitema, 2020-10-11:
I have exactly the same issues with TMC5161 v1.2 drivers from Bigtreetech. I have 8 drivers and all of them cut out when you set the current over 0.5A as described above. I also bought a single v1.5 TMC5160 from Watterott, this driver also has the same issue. 

I also tried with Bart Drings 4_Axis_ Daisy_SPI CNC controller, and later with a Arduino Mega and the TMC stepper library with some of the example code. I tried many settings but no luck.

So is there any progress on this issue?