teemuatlut/TMCStepper #145: Software SPI, TMC2130 & STM32F103

damienmg, 2020-08-16:
Hi,

Sorry this is not really a  bug report but I don't know how to proceed further.

I am developing a dedicated firmware using a board that I have lying around (SKR E3 DIP with a STM32F103RET6 MCU) along with TMC2130.

Whereas I can control the stepper and get status using the Marlin firmware and using SW SPI, I cannot seems to make it works on my own code.

I then try to use the Software SPI example from the TMC Stepper library, without success (status is always at 0):
```
Start...
DRV_STATUS=0b0
DRV_STATUS=0b0
Dir -> 0
DRV_STATUS=0b0
Dir -> 1
DRV_STATUS=0b0
Dir -> 0
DRV_STATUS=0b0
Dir -> 1
DRV_STATUS=0b0
Dir -> 0
DRV_STATUS=0b0
Dir -> 1
```

(The board is connected to a 24V power supply, the same I got success with Marlin).

I attached [a minimal reproduction](https://github.com/teemuatlut/TMCStepper/files/5081322/test-fw.zip), taking the Marlin build as basis to get it to build (with that setup I can use the screen & sd no issue). The example is switched to Serial2 instead of Serial because the USB serial is a bit finickier to make work.

Anyhow I am at lost and I probably missing something obvious but I don't know where to ask this, any chance you could take a quick look?

teemuatlut, 2020-08-17:
I can try it myself when I have the time but until then make sure your hardware setup is correct. VMOT is applied, VIO is at 3.3V, all the SPI lines are connected to the right pins on both sides.

damienmg, 2020-08-17:
Thanks a lot.

Just to clarify the driver works well with Marlin with the same version of the TMCStepper library. I just switch the software and the new code just get 0 as DRV. I suspect an issue with SPI communication but I could not pinpoint what is my mistake.

I continue to dig in the Marlin source code to try to find a big difference but I could not find one so far.