teemuatlut/TMCStepper #153: No working Example(TMC2130)

wolfgangmauer, 2020-09-23:
I use a breadboard with Arduino UNO and a BTT TMC2130 V3.0 and try to get any of the examples running.
NONE of the existing Example seems to work(Compile) out of the Box. (Of cource with the right driver enabled)

Is there any?

teemuatlut, 2020-10-04:
For compile issues it helps to post the compile error.
To get the driver itself working, first make sure your SPI communication works properly.

wolfgangmauer, 2020-10-04:
@teemuatlut  I was trying to build a shared library and had some problems, that was my mistake.
Is there a "easy" way to check if the SPI work communication properly?
The Pi has hardware spi but with the "Simple.ino" the Motor do not move.

teemuatlut, 2020-10-04:
Read back IOIN or more specifically the version value as that should never be 0 or FF.
For porting communication to a new platform or a new MCU, even a cheap logic analyzer is a valuable tool to have.

wolfgangmauer, 2020-10-04:
@teemuatlut  THX!