teemuatlut/TMCStepper #154: TMC2209 V2.1

Mriwaa, 2020-09-23:
Hi , i am working on a project and i want to use the tmc2209 to control a stepper motor . But there is not much documentation on the internet and i don't know how to connect it to arduino or what code should i whrite to change the features of the driver . 
Can someone please help me , i really don't know from where to start

wolfgangmauer, 2020-09-24:
With the TMC2208/2209, features must be set via UART as shown here
[image]
And the Simple.ino shows what & how to use 
Uncomment
//TMC2209Stepper driver(&SERIAL_PORT, R_SENSE, DRIVER_ADDRESS);

Mriwaa, 2020-09-24:
thanks for your response , but the problem here is that the version v2.1 has  two UART pins : TX and RX pins , that's why i can't do it as shown on the picture . Should i just connect TX to RX  pin of the arduino and vice versa ?

wolfgangmauer, 2020-09-24:
You should be trying a little ;-)
https://wiki.fysetc.com/Silent2209/#v21-hardware-connection