teemuatlut/TMCStepper #156: TMC2226 Marlin 2.0.6.1

cobaltduff, 2020-10-01:
TMC2226 is configured to #define X_RSENSE 0.11 or #define X_RSENSE 0.15 ? is there any more change? Thank you

teemuatlut, 2020-10-03:
For 2226 the driver type 2209 should work as they're the same IC in a different package. The sense resistor values depend on your hardware. Make sure to check the serial addresses.

IJeffray, 2020-11-22:
I get no response from teh 2226 using the 2209 driver type.  The 2208 driver type works, but I only see very low current at the motor, no matter the settings.   I'm using the Trinamic 2226 EVAL board with not much luck.

W-M-D, 2021-11-17:
I'm having the same issue with the 2209 driver. haven't tried the 2208 driver yet. No uart response. It does seem to change the settings because after begin() the motor engages. but it doesn't move