teemuatlut/TMCStepper #157: 5160 FULL SPI mode

luiscarone, 2020-10-11:
Hi there, so far I was not able to run TMC 5160 bigtree 1.2 using rampmode, full SPI control. I'm using a 5160 with esp32, with a 3A stepper nema 23, I could achieve a High torque with normal STEP/DIR pins, but the motor stall with a TSTEP (20), under Spread Cycle mode, using this settings; (my max RPM is around 800rpm.)
 driver.toff(1);
  driver.blank_time(24);
  driver.tbl(1); 
  driver.tpfd(0);
  driver.disfdcc(1);  
  driver.irun(31);
  driver.hend(12);
  driver.small_hysteresis(0);
  driver.microsteps(16);

I connect different step drivers even some cheap stepsticks with lower current, and was able to get more torque at high rpm. using Full Step mode I always stall at TSTEP 30.. so I was trying to run with DC Step using the motion controller, but was not able to make the motor spin using VMAX and RAMPMODE 1. 

There is a better settings for high rpm? or this is a limitation from the drive. I know I will always lose torque at High RPM, but since i'm able to achieve this with others drivers I don't wanna give up so I'm crazy here trying to understand where is the limitation. (probably in my code...)

teemuatlut, 2020-10-11:
One of the more important things that can affect the performance in higher speeds is your voltage level.

A high hend value can limit your chopper frequency.
You don't need to call both blank_time and tbl as those do the same thing.

Honestly, to properly fine tune the chopper waveform you'd greatly benefit from inspecting with an oscilloscope. Until then it's mostly educated guesswork.
You could also try asking Trinamic directly about which settings to use for high speed applications. They're the real experts after all.

luiscarone, 2020-10-11:
Thanks Teemuatlut, I manage to get higher RPM with a lower hend, thanks, and with lower microsteps, rpm skyrocket but I really don't want to lose the precision of at least 16 microsteps.. 

I also play dedge and got more rpm, but the motion was really crappy at low rpm... I manage to get an oscilloscope, but I don't have much experience reading the values, I could see the steps changing in time, even after the stall of the motor, but I don't have the knowledge to understand much more than that... 

I was trying to use Steathchoop, but there was always a bump in TPWMTHRS, and since I'm trying to find RPM and Torque now, I didn't play around much...

luiscarone, 2020-10-11:
and I tried with 24V an 32V as well..

CARMICH, 2020-11-02:
Very new to the game.
I too have been trying to use a 5160 in full SPI mode. I have a SilentStepStickTMC5160 and a TMC5160 BOB.

For the past week I have been trying to do so with circuitpython. I can send all the bytearrays commands over SPI (verified with a logic analyzer). I tried many different modes, including the initialization example in the datasheet. 

I can switch to C++ with a bit of brushing up, but what am I missing? Shouldn't it work?

teemuatlut, 2020-11-02:
I've started with an example to show how to control the 5130/5160 with the integrated motion controller using just SPI commands. Once I've verified that it works and has all the checks in place that I want to include, you can copy the required commands from there.
The basic procedure would be to check your hardware is set up correctly. SD_MODE is set for ramp generator and SPI_MODE for SPI. Then set off time and the current settings (irun. ihold and global scaler). Finally set  ramp mode for velocity (1 or 2) and then you should be able to start sending vmax commands to spin the motor. There may be other steps that I'm forgetting right now but that'll become apparent when I work on the example.

CARMICH, 2020-11-03:
Awesome,  I'll keep checking back. In the meantime I'll mess around with it. Thanks

teemuatlut, 2020-11-04:
```cpp
#include <Arduino.h>
#include <TMCStepper.h>

constexpr uint16_t csPin = 10;
//constexpr uint16_t enPin = 2;
constexpr float r_sense = 0.075;

void setup() {
	delay(250);
	Serial.begin(115200);
	while(!Serial);
	Serial.println("\nTMCStepper TMC internal motion controller example.\nCompiled " __DATE__ " " __TIME__);
	Serial.println("Start...");

	pinMode(csPin, OUTPUT);
	//pinMode(enPin, OUTPUT);
	digitalWrite(csPin, HIGH);
	//digitalWrite(enPin, LOW);

	TMC5160Stepper driver(SPI, csPin, r_sense);
	SPI.begin();

	TMC5160_n::IOIN_i<TMC5160Stepper>::IOIN_t ioin{driver.IOIN()};

	if (ioin.version == 0xFF || ioin.version == 0) {
		Serial.println("Driver communication error");
		while(true);
	}

	Serial.print("Driver firmware version: ");
	Serial.println(ioin.version);

	if (ioin.sd_mode) {
		Serial.println("Driver is hardware configured for Step & Dir mode");
		while(true);
	}
	if (ioin.drv_enn) {
		Serial.println("Driver is not hardware enabled");
		while(true);
	}

	driver.toff(3);
	driver.rms_current(800);
	driver.en_pwm_mode(true);

	driver.A1(1000);
	driver.V1(50000);
	driver.AMAX(500);
	driver.VMAX(200000);
	driver.DMAX(700);
	driver.D1(1400);
	driver.VSTOP(10);
	driver.RAMPMODE(0);
	driver.XTARGET(-51200);

	while(true) {
		delay(1000);

		auto xactual = driver.XACTUAL();
		auto xtarget = driver.XTARGET();

		char buffer[256];
		sprintf(buffer, "ioin=%#-10lx xactual=%7ld\n",
			driver.IOIN(),
			xactual
			);
		Serial.print(buffer);

		if (xactual == xtarget) {
			driver.XTARGET(-xactual);
		}
	}
}

void loop() {}
```
[image]
Arduino NANO 33 IOT
Step, Dir and En were hardwired to ground. Other than that it was a typical setup of connecting SPI and providing the necessary voltages. SD_MODE bridge was cut, setting the driver to use internal motion controller.

I used the TMC2300 branch, but note that the call to the constructor needs to be in local scope (as in; not global) because currently in the newer branches there are SPI calls in the constructor and some platforms can hang because the peripheral is not yet initialized at that point.

CARMICH, 2020-11-05:
Thank you so much, this will help tremendously. I sent you a little
donation the other day, cheers.

Chad

CARMICH, 2020-11-13:
I got everything to work properly with one motor. How would I control more than one TMC5160?

teemuatlut, 2020-11-13:
Create another instance of the driver object with a different Chip Select pin and configure it the same way as the first one.
Each driver should have a unique CS line but the MOSI, MISO and CLK lines can be shared.

CARMICH, 2020-11-13:
Awesome. I got it to work late last night. My issue was one of my tmc5160
was burnt up. Im working through whether I need to send all the settings
each time I switch motors or if I needed to spi.begin and spi.end each time
I switch.
Again, thank you for your help and all the work you have put into this
repository.

Sent from Chad Reynolds mobile.

teemuatlut, 2020-11-15:
You only need to configure the driver once and you don't need to call `spi.begin` or `spi.end` after setup.

coolerooney, 2020-11-17:
Hi Teemualut

first thx for your work on these steppers!

we are duplicating your Nano/5160 stepstick example, but we have difficulty in selecting the full SPI mode
you mention: 

SD_MODE bridge was cut, setting the driver to use internal motion controller.

Question: did you actually cut the bridge? which bridge was cut?

we are using the FYsetc 5160 V1.2, which should default to full SPI mode, the solder bridge as delivered- but still only movement via S/D

[image]

also ordered some 5160s from Trinamic just to be sure

Thx for your help

Best

Coolerooney

teemuatlut, 2020-11-17:
You can read the SD_MODE pin state from the IOIN register like I did in the code above. That will tell you how the driver itself sees it, regardless of any hardware modifications.

The driver I used were Watterott 5160 SIlentStepSticks, version 1.5
https://github.com/watterott/SilentStepStick/blob/master/hardware/SilentStepStick-TMC5160_v15.pdf
I cut the SD_MODE line from that schematic, which then pulls the SD_MODE to ground.

teemuatlut, 2020-11-17:
I'm trying to find a schematic for the FYSETC driver but I'm thinking that the pads you showed control the SPI_MODE line. Meaning that they control if the driver uses SPI for configuration (current, stepping mode, etc) or if it uses HW pins for setting the same parameters. This is **not** the same as selecting between internal ramp generator and step/dir control.

coolerooney, 2020-11-24:
Hi Teemuatlut

we duplicated your set up- but no joy- used trinamic stepstick

you mention:
**Step, Dir and En were hardwired to ground.** - must this be hardwired, or via configuration?

 **SD_MODE bridge was cut, setting the driver to use internal motion controller.**
what did you exactly cut?- where is this bridge? cant find it on the PCB

we can only move it now in S/D, in SPI, the motor locks up- Xactual moves, but not the motor

thx!

best

C

teemuatlut, 2020-11-24:
Step and dir can be left floating but Enable line needs to be pulled LOW to enable the driver.
[image]
Upper pad on the left top corner.
The IOIN register will tell you the pin status.

coolerooney, 2020-11-24:
Hey T

thx for the quick response

still a bit confused by your remark SD Mode Bridge was cut- is this the pad with the slightly different colour inbetween the gold squares?

Sorry for being a Noob- hahahahahha

best

c

teemuatlut, 2020-11-24:
Yes

coolerooney, 2020-11-24:
It works🤓

johnsmakar, 2020-11-25:
Can the example sketch on on the 5161?

johnsmakar, 2020-11-25:
Can the example sketch on the 5161?

johnsmakar, 2020-11-25:
Can the example sketch run on the 5161?

teemuatlut, 2020-11-25:
The 5161 is the same as 5160 but with integrated mosfets so I would expect it to work the same way.

rohan-patel367, 2024-03-08:
Hello, I am seeking clarification on whether it is possible to control a TMC5160 board using SPI completely (ie no STEP/DIR pins). I have tried the example code posted above but it does not work. I'm not sure if my hardware is impacting my ability to get this to work... I am using the FYSETEC QHV5160

draffetto2024, 2024-03-09:
rohan-patel367, to use Full SPI you have to rather absurdly solder bridge the physical SD pin on the TMC driver to ground and remove its 0 ohm resistor to VCC IO. Below is a detailed picture for our TMC5160T. You can follow the datasheet for your driver, find SD and SPI mode pins and trace them on the physical board and then find which resistor to remove and where to solder bridge. The TMC5160 is unfortunately not super well documented, so praise teemuatlat for making this library. 

[image]

rohan-patel367, 2024-05-26:
Hi @teemuatlut, I have gotten my hands on a silent step stick which has the ability to be manually configured to use the internal motion controller... but the example code you wrote no longer works. Are you able to point me in the right direction for which branch/commit i should checkout to get things working?