teemuatlut/TMCStepper #158: Motion controller mode

jarpekar, 2020-10-23:
Hello , did somebody tried to use this library as motion controller mode. 
Meaning that sending speed and destination via SPI and chip has full control motor ?

thanks a lot.

teemuatlut, 2020-10-23:
Take a look at chapter 22 in the datasheet.

jarpekar, 2020-10-23:
Uhmm wich datasheet do you exactly mean. Do you mean trinamic?

teemuatlut, 2020-10-23:
Yes, the TMC5160 datasheet.

jarpekar, 2020-10-23:
Yeah we already had working code with TMC. Just asking if this library support this register for target position and acc and velocity. Just trying to share experiencis but we very like your code. Nice clean and well structured.

teemuatlut, 2020-10-23:
You can see the interface in the doxygen documentation here: https://teemuatlut.github.io/TMCStepper/class_t_m_c5160_stepper.html
Just be sure to expand inheritance from 5130 as many of the motion controller methods are derived from there.

jarpekar, 2020-10-23:
Thanks for direction, i will share my modbusTCP/IP inplementation  here also with kicad drawing. Hope it will be usefull.

jarpekar, 2020-10-26:
Hello is there  any function on tmc 5130 that will easy show is my spi comunication is OK? We got ethernet shield on spi aswell. Thanks a lot for help.

teemuatlut, 2020-10-27:
Read back the version number in the IOIN register. It should likely read 0x30 for the 5160 but at least it should not read 0 or 0xFF.

jarpekar, 2020-11-01:
Hello can anyone help ?  Using this settup. sorry for description. Czech language. Basicli UNOV3 with Ethernet shield 2 

[image]
 ---- 
 
Looks like I get some communication running. But atc wierd. I got running code with setup  SPI.beginTransaction(SPISettings(4000000, MSBFIRST, SPI_MODE3));
it match parametres in TMC2130. 

 THANKS a lot for help. 

terminal log :
Start...
DRV_STATUS=0b0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:15
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0     
Driver status:0
X actual:0     
Driver status:0
X actual:0     
Driver status:0
X actual:0     
Driver status:0
X actual:1946157056
Driver status:1001010000010000000000000000000
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:11111010000000010000
X actual:15747592
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:0
Driver status:0
X actual:-131072

CODE :
> 
#include <Arduino.h>
#include <TMCStepper.h>
#include <MgsModbus.h>
#include <Ethernet.h>

// pinout Pro motor
#define CSTETH2         10 // chip select for eth drive
#define EN_PIN           7 // Enable motor
#define CS_PIN           8 // Chip select for motor
#define SW_MOSI          12 // Software Master Out Slave In (MOSI)
#define SW_MISO          11 // Software Master In Slave Out (MISO)
#define SW_SCK           13 // Software Slave Clock (SCK)

//#define SW_RX            63 // TMC2208/TMC2224 SoftwareSerial receive pin
//#define SW_TX            40 // TMC2208/TMC2224 SoftwareSerial transmit pin
//#define SERIAL_PORT Serial1 // TMC2208/TMC2224 HardwareSerial port
//#define DRIVER_ADDRESS 0b00 // TMC2209 Driver address according to MS1 and MS2

#define R_SENSE 0.11 // Match to your driver
                      // SilentStepStick series use 0.11
                      // UltiMachine Einsy and Archim2 boards use 0.2
                      // Panucatt BSD2660 uses 0.1
                      // Watterott TMC5160 uses 0.075

// Select your stepper driver type
//TMC2130Stepper driver(CS_PIN, R_SENSE);                           // Hardware SPI
//TMC2130Stepper driver(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK); // Software SPI
//TMC2660Stepper driver(CS_PIN, R_SENSE);                           // Hardware SPI
//TMC2660Stepper driver(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK);
TMC5160Stepper driver(CS_PIN, R_SENSE);
//TMC5160Stepper driver(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK);

//TMC2208Stepper driver(&SERIAL_PORT, R_SENSE);                     // Hardware Serial
//TMC2208Stepper driver(SW_RX, SW_TX, R_SENSE);                     // Software serial
//TMC2209Stepper driver(&SERIAL_PORT, R_SENSE, DRIVER_ADDRESS);
//TMC2209Stepper driver(SW_RX, SW_TX, R_SENSE, DRIVER_ADDRESS);

void setup() {
  // ************  Set communications
  Serial.begin(9600);
/*   Serial.println("Serial interface started");
  Ethernet.begin(mac,ip,gateway,subnet);
  server.begin();
  Serial.println(Ethernet.localIP()); */

  //************* Set TMC motor and SPI
  pinMode(CSTETH2,OUTPUT);
  pinMode(EN_PIN,OUTPUT);
  // Enable driver in hardware

  //DRIVER SETUP
                                  // Enable one according to your setup
                                  // SPI drivers
  SPI.begin();  
	while(!Serial);
	Serial.println("Start...");
	driver.begin(); 			// Initiate pins and registeries
	driver.rms_current(600); 	// Set stepper current to 600mA. The command is the same as command TMC2130.setCurrent(600, 0.11, 0.5);
	driver.en_pwm_mode(1);  	// Enable extremely quiet stepping
	Serial.print("DRV_STATUS=0b");
	Serial.println(driver.DRV_STATUS(), BIN);

}

bool shaft = false;

void loop(){
// Run 5000 steps and switch direction in software
// Serial.print("Test connection: ");
// Serial.println(driver.test_connection());
Serial.print("X actual:");
Serial.println(driver.XACTUAL());
Serial.print("Driver status:");
Serial.println(driver.DRV_STATUS(),BIN);
driver.XTARGET(500);

delay(1000);
driver.XTARGET(0);

if (driver.XACTUAL() == 500){
driver.XTARGET(500);
};

}

teemuatlut, 2020-11-01:
Based on your schematic it looks like you're using the 5160 SilentStepSticks, which are hardware configured for STEP/DIR control. See the SD_MODE pin.

When you overcome that, first get the driver communication working without the ethernet shield.

jarpekar, 2020-11-01:
Yeah you are right, We cut the SD_Mode . this is not a problem anymore. Same setup im using is alive and working on other code just write Targer register. 

I just want to use this lib to have comfortable way fo configure stepper etc. 
I think there is some communication mess in SPI bewtheen Eth shiels and driver. 

Is there any way to debug spi ?

teemuatlut, 2020-11-01:
The best way would be with a logic analyzer.

You can also try manually sending the datagrams to read back IOIN register.
Or you can plug into the library methods that are defined as "weak". That way you can inject your own serial debugging output. The 5160 uses the SPI methods from 2130: https://github.com/teemuatlut/TMCStepper/blob/e3a59778d50333b53513c248e5a1de7784731db0/src/source/TMC2130Stepper.cpp#L57-L162

teemuatlut, 2020-11-04:
Please see a working example of the motion controller here in another thread: https://github.com/teemuatlut/TMCStepper/issues/157#issuecomment-721960004

jarpekar, 2020-11-04:
Thanks a lot for help, issue is solved. Problem was with spi. Ethernet library keeps  CS pin for ethernet shield low.  So when spi comms drop pin low there were data mischmash.

Right now I try to get some working setuo for Nema 23 stepper. 

When i use calculation sheet  its not working for me... Im not that much expert in motion controllers and motors. 

 Thanks a lot for help.  I will share  working code when its tested properly.

Thanks for help.