teemuatlut/TMCStepper #159: TMC2209: Do I have to connect serial RX?

MarcelRobitaille, 2020-10-30:
I am looking at the examples, the code, and the datasheet for the IC, and I am confused when I have to connect the RX pin and when not.
I assume anytime we have to read data, we need the setup on the left. But I am not sure exactly what methods require to read data. Would it be possible to have a list of functionality that doesn't require reading? I would be willing to write some more complete documentation if someone could give a brief description.

[image]
https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V105.pdf

teemuatlut, 2020-11-02:
For HW serial you always need to connect both.
For SW serial it can depend on the platform. Since your muxing the RX and TX lines, the distinction becomes a bit blurry but on AVR the pin still needs to be interrupt capable.

The registers that do not need read capabilities are listed as write only in the datasheet.

ThomasLengeling, 2021-01-28:
Thank you!, I am having problems receiving data from the TMC2209 on the RX. I connected the RX and TX as the figure that @MarcelRobitaille  shared with you.  The TX line is working perfectly and I am able to communicate and control the TMC2209 without a problem. However I am not having any response.

I am using the following settings on the teensy 4.0. The TX and RX are connected to the Serial3 HR Port and there is 1K resistor in between the RX and TX lines. 

```

#define SERIAL_PORT Serial2
#define DRIVER_ADDRESS 0b00

#define R_SENSE 0.11f

TMC2209Stepper driver(&SERIAL_PORT, R_SENSE, DRIVER_ADDRESS);

void loop(){
  Serial.begin(115200);
  delay(1000);
  Serial.println("starting motors");

  pinMode(EN_PIN_01, OUTPUT);
  pinMode(STEP_PIN_01, OUTPUT);
  pinMode(DIR_PIN_01, OUTPUT);
  digitalWrite(EN_PIN_01, LOW);      // Enable driver in hardware

  SERIAL_PORT.begin(115200);      // HW UART drivers

  driver.begin();               
  driver.toff(5);                
  driver.rms_current(300);      
  driver.microsteps(32);         

  driver.en_spreadCycle(false);  
  driver.pwm_autoscale(true);   

  Serial.print(F("\nTesting connection... 0"));
  uint8_t result = driver.test_connection();
   delay(500);
  if (result) {
    Serial.println(F("failed!"));
    Serial.print(F("Likely cause: "));
    switch (result) {
      case 1: Serial.println(F("loose connection")); break;
      case 2: Serial.println(F("Likely cause: no power")); break;
    }
    Serial.println(F("Fix the problem and reset board."));
    // abort();
  }
  Serial.println(F("OK"));

}
```

ThomasLengeling, 2021-02-01:
sorry for the msg, it's working now, thanks for the great lib!