teemuatlut/TMCStepper #160: MKS Gen L v1.0 + TMC2209 V3 UART

Koby88, 2020-11-12:
Hi all i have problem and i dont know if it is software related or hardware... I hope you can help me out.
Recently i bought MKS Gen L v1.0 board with 5 TMC2209 drivers for a cheap. Problem is i cant get them work vith UART.
The drivers are a little strange, on all TMC2209 that i found on tutorials there are 3 pins, but on mine only 2.
Drivers are fysetc 2209 v3. Here is a link [wiki.fysetc.com]
As a hobbyist that schematics wont really help me coz i dont understand it...
I tried all i could, solder 1k resistor on one pin, tried with two separate wires one on tx one on rx pin, tried to swap plases, tried to remap dedicated pins, #define SOFTWARE_DRIVER_ENABLE on and off, ... no luck
Here is my configuration and some pictures. Pls help i really ned printer to work thanks in advance
https://ibb.co/1bZ9gtD
https://ibb.co/GdJ8MhN

teemuatlut, 2020-11-13:
I don't have any MKS boards myself but looking at the schematic below, it seems like FYSETC integrated the TX lines resistor into the driver and you should just connect the Serial lines individually.
[image]

SOFTWARE_DRIVER_ENABLE was developed with BSD2660 in mind as it does not have a dedicated Enable pin and the driver couldn't be disabled without a software write. You shouldn't use it with these drivers.

Koby88, 2020-11-13:
Yeah thats what i think i connect TX and RX pins from driver to dedicated TX and RX pins on board and in pronterface it gave me a error All low, no mether what i do... Could bi firmware related ?

ThomasLengeling, 2021-01-29:
@Koby88 Did you manage to solve this issue, I am having similar problems with the MKS 2.0, I connect both TX and RX to the Serial3 of teensy 4.0, could it be that there is something with the firmware that RX is not communicating? Thank you in advance!