teemuatlut/TMCStepper #161: sketch/src/MotionControl.cpp:119:74: error: too many initializers for 'float [3]'

prak1111, 2020-11-15:
Hello I am trying to compile 3 axis on latest version. Arduno IDE is latest release.  But getting error like -
WARNING: library TMCStepper claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
sketch/src/MotionControl.cpp: In function 'void mc_arc(float*, plan_line_data_t*, float*, float*, float, uint8_t, uint8_t, uint8_t, uint8_t)':
sketch/src/MotionControl.cpp:119:74: error: too many initializers for 'float [3]'
     float previous_position[MAX_N_AXIS] = { 0.0, 0.0, 0.0, 0.0, 0.0, 0.0 };
                                                                          ^
sketch/src/Limits.cpp:256:131: error: too many initializers for 'uint8_t [3][2] {aka unsigned char [3][2]}'
                                       { A_LIMIT_PIN, A2_LIMIT_PIN }, { B_LIMIT_PIN, B2_LIMIT_PIN }, { C_LIMIT_PIN, C2_LIMIT_PIN } };
                                                                                                                                   ^
Multiple libraries were found for "SD.h"
 Used: /home/prak/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SD
 Not used: /home/prak/Downloads/arduino-1.8.13-linux64/arduino-1.8.13/libraries/SD
Multiple libraries were found for "WiFi.h"
 Used: /home/prak/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 Not used: /home/prak/Downloads/arduino-1.8.13-linux64/arduino-1.8.13/libraries/WiFi
exit status 1
Error compiling for board ESP32 Dev Module.

prak1111, 2020-11-15:
After making max axis to 6, above error vanished but as i am running ubuntu and have python3  instead of python. So it is throwing error like -

WARNING: library TMCStepper claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
Multiple libraries were found for "WiFi.h"
 Used: /home/prak/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 Not used: /home/prak/Downloads/arduino-1.8.13-linux64/arduino-1.8.13/libraries/WiFi
Multiple libraries were found for "SD.h"
 Used: /home/prak/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SD
 Not used: /home/prak/Downloads/arduino-1.8.13-linux64/arduino-1.8.13/libraries/SD
exec: "python": executable file not found in $PATH
Error compiling for board ESP32 Dev Module.

teemuatlut, 2020-11-15:
Only the first line is relevant to this library and even that is merely a *warning* (not an error) because I don't officially claim to support ESP32 even if it has been proven to work.

johnsmakar, 2020-11-26:
Do you have plans in the foreseeable future to support the ESP32? There seems to a bit of interest around it; myself included.

teemuatlut, 2020-11-26:
The library has been proven to work with ESP and the SPI drivers. I would expect HW Serial to work as well.
For SW Serial the framework supposedly [can map the serial port to any logical pins](https://circuits4you.com/2018/12/31/esp32-hardware-serial2-example/).

johnsmakar, 2020-11-26:
In regards to the 5160/5161 where SPI is the only communication available, I get compilation errors such as:
(using the TMCStepper-tmc2300 branch)

TMC_HAL_Arduino.cpp:37:40: error: invalid conversion from 'char*' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
         TMC_HW_SPI->transfer(buf, count);
                                        ^

SPI.h:68:10: note:   initializing argument 1 of 'void SPIClass::transfer(uint8_t*, uint32_t)'
     void transfer(uint8_t * data, uint32_t size);
          ^

TMC_HAL_Arduino.cpp:61:40: error: invalid conversion from 'char*' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
         TMC_HW_SPI->transfer(buf, count);
                                        ^

SPI.h:68:10: note:   initializing argument 1 of 'void SPIClass::transfer(uint8_t*, uint32_t)'
     void transfer(uint8_t * data, uint32_t size);
          ^

what would be the proper approach to fix this?

teemuatlut, 2020-11-27:
The 2300 branch is under development but the problem comes from the Arduino framework not having a consistent interface across all platforms. You can cast the passed values to fix it until I get around to working with this again.

johnsmakar, 2020-11-28:
sorry, cast them from where?
In the sketch where I am creating an instance of the driver...
`TMC5160Stepper driver(SPI, csPin, r_sense);`
or in the TMCStepper.h file?
`TMC5160Stepper(SPIClass &spi, TMCStepper_n::PinDef pinCS, float RS, int8_t link_index = -1);`

teemuatlut, 2020-11-29:
`TMC_HAL_Arduino.cpp`
```diff
void TMC_SPI::transfer(char *buf, const uint8_t count) {
    if(TMC_HW_SPI != nullptr) {
-        TMC_HW_SPI->transfer(buf, count);
+        TMC_HW_SPI->transfer(reinterpret_cast<uint8_t*>(buf), count);
    }
}

void TMC2660Stepper::transfer(char *buf, const uint8_t count) {
    if(TMC_HW_SPI != nullptr) {
-        TMC_HW_SPI->transfer(buf, count);
+        TMC_HW_SPI->transfer(reinterpret_cast<uint8_t*>(buf), count);
```