teemuatlut/TMCStepper #165: TMC5160 consumes more current than configured, power supply is turned off with overcurrent protection

fenixil, 2021-01-07:
I use Bigtreetech TMC5160 v1.2 driver in the full SPI mode (SPI_MODE - high, SD_MODE - low). 
I set RMS current to 750mA however when there is a load on the stepper motor, driver sucks more current from the power supply and it is turned off with overcurrent protection. I noticed that the current can spike up to 2A when the motor stall. I see the same behavior for both StealthChop and SpreadCycle modes.

Why current exceeds configured RMS?
Is there a way to limit the current so that my power supply is not turned off?
How can I calculate max current which can be consumed by the motor?

Code for back and forth rotation of the steppers, just apply the load on the motor and monitor the current:
```
#include <TMCStepper.h> // https://github.com/teemuatlut/TMCStepper

#define MICROSTEPS 16

class DriverConfig {
private:
  constexpr static int32_t min_target_position = 100 * MICROSTEPS;

  Stream * serial;
  int32_t  target_position = min_target_position * 2;

public:
  DriverConfig(): DriverConfig(&Serial) {}
  DriverConfig(Stream * logger) { this->serial = logger; }

  uint32_t VSTART = 0;
  uint32_t VSTOP  = 100;

  uint32_t V1   = 5;
  uint32_t A1   = 1;
  uint32_t D1   = 5;
  uint32_t AMAX = 50;
  uint32_t DMAX = 50;
  uint32_t VMAX = 10 * 200 * MICROSTEPS;
  uint32_t RMS_CURRENT = 750; // mA

  int32_t targetPosition() { return target_position; }
  void    targetPosition(int32_t tp) {
    if(tp < min_target_position) {
      target_position = min_target_position;
      serial->print("WARNING: Target position cannot be less than ");
      serial->println(min_target_position);
    } else {
      target_position = tp;
    }
    serial->print("New Target Position:");
    serial->println(target_position / MICROSTEPS);
  }

  void printToSerial() {
    serial->print("{");
    uint32_t arrValues[] =
        {target_position, VSTART, VSTOP, V1, A1, D1, AMAX, DMAX, VMAX, RMS_CURRENT};
    char * arrTitles[] = {"target_position",
                          "VSTART",
                          "VSTOP",
                          "V1",
                          "A1",
                          "D1",
                          "AMAX",
                          "DMAX",
                          "VMAX",
                          "RMS_CURRENT"};
    for(int i = 0; i < sizeof(arrValues) / sizeof(uint32_t); i++) {
      serial->print(arrTitles[i]);
      serial->print(": ");
      serial->print(arrValues[i]);
      serial->print(", ");
    }

    serial->println("}");
[... 172 lines cut ...]
  static bool        forward;

  uint32_t ms = millis();

  // Serial control experiments
  while(Serial.available() > 0) {
    int8_t read_byte = Serial.read();
  }

  if((ms - last_time) > 100) { // run every 0.1s
    RAMP_STAT_t rs{driverLeft.RAMP_STAT()};
    if(rs.position_reached) {
      forward        = !forward;
      auto targetPos = forward ? conf->targetPosition() : 0;
      driverLeft.XTARGET(targetPos);
      // invert the position to rotate in the opposite direction
      driverRight.XTARGET(-targetPos);
      Serial.print("POS: ");
      Serial.println(targetPos);
    }

    last_time = ms;
  }
}
```

daniel-frenkel, 2021-04-03:
RMS current is Root Mean Square. It is the average current used in the moment. 

Max current is RMS x 1.41

Also be sure your sense resistor value is correct as that will affect the current

Antonio-Citati, 2022-04-06:
Just a side question: is there any way to read the motor current from the TMC5160? Since it can limit the current allowed to the motor I guess there must be a measurement of it somewhere. Is this information somehow accessible?

daniel-frenkel, 2022-04-06:
You can only read back the current that you set, you can’t monitor it in real time. I use my power supply to monitor current when testing things out. 

Is there a reason you need to keep track of the current? 

Remember that the current you set is the RMS current. It’s just an average and can max out at 1.41 times the RMS current that you set. The current is constantly changing depending on the load, speed, and acceleration.

Antonio-Citati, 2022-04-07:
Thank you for the reply, I know it is an RMS and that current always changes.  I run with 2 TMC5160 the 2 motors of a totally automatic telescope on a gondola attached to a weather balloon at around 35km altitude and need to monitor power consumption closely as a way to optimize battery usage, and to address any issue which could lead to spikes in motor power consumption. Will just design circuit with additional 2 more current sensors. We have quite a few already and just wondered if these could be substituted by a current measurement function on the chip.

daniel-frenkel, 2022-04-07:
That project sounds incredible. Do you have more info on it so I can learn more?

issuefiler, 2023-09-16:
> Just a side question: is there any way to read the motor current from the TMC5160? Since it can limit the current allowed to the motor I guess there must be a measurement of it somewhere. Is this information somehow accessible?

`CS_ACTUAL`.