teemuatlut/TMCStepper #167: TMC5160 SPI Clear Stallguard Indicator / Register

itsyoshio, 2021-01-14:
Hi there,

i'm having a difficult time in trying to clear a FLAG.
The Motor is set up to Stop on Stall `drv.sg_stop(1);` and does so very well.

I can confirm this by reading `drv.sg_result();` and `drv.stallguard();`

Now, the Datasheet tells me i should write a `1` to `event_stop_sg` to Clear the Stallguard Indicator
[image]
but that isn't possible in [RAMP_STAT](https://github.com/teemuatlut/TMCStepper/blob/11aefc1ba6cef2a078944f05eaaf734aa7964996/src/source/RAMP_STAT.cpp#L16) as there is no **SET**ter

There isn't one in DRV_STATUS either.

How would i go about on clearing (StallGuard) Indicators?

Thanks! <3

fenixil, 2021-01-26:
I had the same issue and managed to reset the stall status as described in the datasheet. Unfortunately write function is or protected, so I had to derive my stepper class from the one defined in the library and call the write method:
```
class TMC5160StepperExt : public TMC5160Stepper {
public:
  TMC5160StepperExt(uint16_t pinCS): TMC5160Stepper(pinCS, default_RS, -1){};
  void resetStall() { write(RAMP_STAT_t::address, RAMP_STAT()); }
};
```

This is the hacky solution, ideally, this method should be supported by the library.

teemuatlut, 2021-01-26:
> this method should be supported by the library

`driver.RAMP_STAT(1<<6);`

teemuatlut, 2021-01-26:
The next release will add the write method but I have no concrete timeline about the release date.

https://github.com/teemuatlut/TMCStepper/blob/586ff2459e7c03c131a3e1d6f28be6b45a508d45/src/source/interfaces/TMC5130.hpp#L432-L434

itsyoshio, 2021-01-26:
Many thanks to both of you.
I'll use your Workarounk for now @fenixil!

closed 2022-03-15 by itsyoshio

vsi5004, 2023-10-22:
Is there any chance that this change has been implemented? I'm currently running into the same issue and would be interested in the non-hack solution. Really appreciate all of the work done in getting the library to this point!

> The next release will add the write method but I have no concrete timeline about the release date.
> 
> https://github.com/teemuatlut/TMCStepper/blob/586ff2459e7c03c131a3e1d6f28be6b45a508d45/src/source/interfaces/TMC5130.hpp#L432-L434