teemuatlut/TMCStepper #168: Trying to understand how to use stallguard

bpstark, 2021-01-18:
I am trying to setup an esp32 to run 2 TMC5160 with stall guard. There seems to be several ways in which to check for a stall 

`driver.stallguard()
`
or
`
TMC2130_n::DRV_STATUS_t drv_status{0};
drv_status.sr = driver_.DRV_STATUS();
drv_status.stallGuard;
`
Which is correct? Do I need to set `diag0_stall` and or `diag1_stall`?
Since I have 2 drivers running on the same SPI bus do I need to manaully call `driver.switchCSpin` before a call?

teemuatlut, 2021-01-18:
The first is implemented like the second.
Making an object however can be useful if you also want to inspect another value in the register without making another SPI call.

`diag#_stall` is used to assign a physical pin to notify of stallGuard status. You don't need it if you're not using the pin.

The library handles CS pin so you don't have to.

bpstark, 2021-01-27:
@teemuatlut thanks for the info. I was able to get most of what I wanted working. It turns out my biggest issue was that I was trying to use both cores of the ESP32. While I was only calling the driver from one core, and using the other core to keep steps, this seemed to cause an issue. It seems that somewhere either in the driver or the underlying SPI code, If you put something into its own task then we get undefined behavior.