teemuatlut/TMCStepper #171: TMC2209 stallguard example question

MakerLS, 2021-01-26:
Hi. Sorry for my bad English.

Thank you for your great work, teemuatlut.
I need help, I don´t understand the line:

#define DRIVER_ADDRESS 0b00 // TMC2209 Driver address according to MS1 and MS2

In the few examples I have been able to find, MS1 and MS2 are wireless I think that for the this line:

driver.microsteps(16);

Makes it unnecessary to configure it by hardware.

Also if each tmc2209 has to have its RX and TX according to documentation, I don't see much sense in defining an address.
I understand that 0b00 is a generic value for DRIVER_ADDRESS, if I'm wrong, how is this value set?
I have checked the definition of the class, and I can't see where it is used.

Thank you

teemuatlut, 2021-01-26:
When the TMC2209 is configured with UART, the MS1 and MS2 pins dictate the address instead of the microstep setting.

If each of your drivers has its own serial line, then all drivers can be set to have address 0.
Alternatively (preferably) you would use just one serial line (use hardware serial if you can) and different addresses for drivers.

The address value is passed to the constructor and each driver *within the same serial line* should have a different address to differentiate each one.

MakerLS, 2021-01-26:
Hi. 
I think I have understood. A single serial port that connects to all tmc2209. Each tmc2209 must have a different address (0b00, 0b01 and 0b02, for example).

The configuration pins M0 and M1 without physically connecting to 5v or GND.

it is right?
thanks

teemuatlut, 2021-01-26:
The MS pins have internal pull down resistors so if you don't connect them to anything, it will default to 0.
To set address 1, you need to connect MS1 to VIO. Basically the addressing follows the binary number pattern.

MakerLS, 2021-01-27:
Okay. So in addition to hardware addressing, software addressing is also necessary.
For example:
TMC2209 "1"
HARDWARE: MS1 = LOW MS2=LOW
SOFTWARE: 0b00 >> DRIVER_ADRESS instead of 0b00 could it just be 0000?

TMC2209 "2"
HARDWARE: MS1 = HIGH MS2=LOW
SOFTWARE: 0b01 >> DRIVER_ADRESS instead of 0b01 could it just be 0001?

TMC2209 "3"
HARDWARE: MS1 = LOW MS2=HIGH
SOFTWARE: 0b02 >> DRIVER_ADRESS instead of 0b02 could it just be 0002?

Sorry if what I ask is too basic, I have not located much information, what little I have found has to do with Marlin.
Thanks!

teemuatlut, 2021-01-27:
`0b` means the number is given in binary form so you can only use ones and zeros.

MS2 | MS1 | DRIVER_ADDRESS
---|---|---
L | L | 0b00 == 0
L | H | 0b01 == 1
H | L | 0b10 == 2
H | H | 0b11 == 3

MakerLS, 2021-01-27:
Now that I've finally found out, I feel pretty stupid hahahaha.
Thank you very much for your work and for your patience