teemuatlut/TMCStepper #172: Library does not compile on Teensy

julian-weinert, 2021-01-27:
Unfortunately I don't have an Arduino to hand to confirm if this issue applies to Arduino as well, but I see no reason for a possible difference.

I'm trying to use the library with a TMC2130 but cannot even compile the "Simple" example.

```
libraries/TMCStepper/src/source/TMC2208Stepper.cpp: In member function 'void TMC2208Stepper::beginSerial(uint32_t)':
/Users/jw/Development/Hardware/Arduino/libraries/TMCStepper/src/source/TMC2208Stepper.cpp:39:14: error: 'class SoftwareSerial' has no member named 'stopListening'; did you mean 'isListening'?
    SWSerial->stopListening();
              ^~~~~~~~~~~~~
              isListening
libraries/TMCStepper/src/source/TMC2208Stepper.cpp: In member function 'void TMC2208Stepper::postReadCommunication()':
libraries/TMCStepper/src/source/TMC2208Stepper.cpp:182:14: error: 'class SoftwareSerial' has no member named 'stopListening'; did you mean 'isListening'?
    SWSerial->stopListening();
              ^~~~~~~~~~~~~
              isListening
```

I don't like issues with not a lot of details, but I have no idea what to add to the log output

julian-weinert, 2021-01-27:
I just tried to patch out SoftwareSerial by setting the `SW_CAPABLE_PLATFORM` switch to false in any case in `TMCStepper.h`
That only leads to more issues of undefined pin names `TMC2208Stepper.cpp:208:7: error: 'RXTX_pin' was not declared in this scope`

Is this library capable to work without SoftwareSerial?
And do I really need to bloat up my binary with unnecessary libraries, or is it possible to only build whats needed?

julian-weinert, 2021-01-27:
Ok, I got it working now. I had to hardcode `#define SW_CAPABLE_PLATFORM false` and then also `#define RXTX_pin 0` at the same location (TMCStepper.h:19).

I'm pretty sure it's not supposed to... "Work" like that, can somebody enlighten me?

julian-weinert, 2021-01-27:
I now got it to compile. It still does not run.
How is this library supposed to work? I sound like a total noob here, but your examples don't work, there is no documentation (except a copy of the header files in Doxygen style, which contains endless dead links) and no explanations or guides.

My stepper now does nothing. It gets holding power and that's it.
After a few seconds my Teensy completely locks up at this line (line 117 in your StallGuard example).
```
DRV_STATUS_t drv_status{0};
    drv_status.sr = driver.DRV_STATUS();
```
No further serial transmissions happen.

Did anybody ever use this library successfully with a 2130 and an actual motor?
I feel a bit mean and / or stupid, but I'm pretty sure an example should work as is.

I'm really sorry for all these comments but since I switched to this library no matter what I do nothing works and if I fix (by patching by local copy of the library) something else falls apart and it works less than before.
When I go back to `TMC2130Stepper` everything works fine.

What exactly is needed to reproduce the following working functionality with the new library?
```
void setup() {
    driver.begin(); // Initiate pins and registeries
    driver.SilentStepStick2130(20); // Set stepper current to 600mA
    driver.stealthChop(1); // Enable extremely quiet stepping
}
void loop() {
    digitalWrite(PIN_STP, HIGH);
    delayMicroseconds(7);
    digitalWrite(PIN_STP, LOW);
    delayMicroseconds(7);
    // ...
    if (dir) {
        driver.shaft_dir(0);
    }
    else {
      driver.shaft_dir(1);
    }
```

teemuatlut, 2021-01-27:
Which Teensy are you using?

julian-weinert, 2021-01-27:
Damn, sorry the one detail I could've given you... It's a 2.0++

teemuatlut, 2021-01-27:
Try https://github.com/teemuatlut/TMCStepper/commit/6f3f0d09b6fcf894277671c8bcc575bf714aa8ca

The Doxygen links should work fine if you open the index.html file locally. It's only because the files are hosted in Github that they appear broken. I'd either need to switch the hosting solution or contact GH about the issue.

julian-weinert, 2021-01-28:
Thanks. Yes, that builds now. But I run into the same issue I described in my [last comment ](https://github.com/teemuatlut/TMCStepper/issues/172#issuecomment-767985079).

Could you quickly confirm if my snipped should work, then I'd open a new issue, or if I got something terribly wrong?

teemuatlut, 2021-01-28:
The most basic functionality should be achievable with the [Simple example](https://github.com/teemuatlut/TMCStepper/blob/master/examples/Simple/Simple.ino).
Remember to uncomment the `SPI.begin` call as you are using the 2130 driver.

julian-weinert, 2021-01-28:
Do I have to keep my patched version or will that go into some update?
Thanks a lot! Also for the reminder, that was my last mistake after succeeding the compilation.

closed 2021-01-28 by julian-weinert