teemuatlut/TMCStepper #173: IRUN vs torque vs rpm

mschorer, 2021-01-29:
I'm using this lib on a digispark pro with oled against 4 x tmc2209s in a grbl machine. It's completely parallel to the grbl controller.
I'm enabling all of the advanced tmc features: hybrid stealth/spread mode, per motor msteps/ihold/irun ... coolstep is on the feature list ...
The lib is working as expected - i'm just trying to tap into your experience ...

I have a setup were is set irun/ihold to 12/4. With this, i'm able to reliably reach the max 1000mm/min @ 25mm/s2.
If i switch to 16/4 - everyting else unchanged - the machine reliably stalls at 1000. It runs at 800.

I would have expected to be able to go faster, because the "critical torque would have struck me later" ...
Any ideas on that?

Setup (see below for exact register values):
- 4 msteps, interpolation on
- external vref disabled - i see the current increase as expected
- stealth to spread handover quite low to take advantage of higher torque in spreadcycle
- 1.2A motors

How do you guys tune the multitude of adjustable parameters? For hybrid?
Is there a good procedure for "most power @ lowest noise"?
Or just trial and error because there are too many factors of mass/resonance/motor parms/ ...
I found the trinamic manuals handle all aspects but mostly in an isolated way ...

```
TmcConfig tmcDefault = { 
    0x1c0,                                // gconf
    ( 4UL | ( 16UL << 8) | ( 2UL << 16)),  // ihold_irun: hold-run-hdelay
    20,                                   // tpowerdown
    90,                                   // tpwmthrs
    0x169102d4,                           // chopconf, 4msteps
    0xc80d0e24,                           // pwmconf
  };
```
BTW, this sounds similar to #99 . Which was never really "resolved" in my eyes ...

teemuatlut, 2021-01-29:
Nice to hear the library is working on a platform I haven't tested it with.

You'll want to look into the Trinamic Application Notes on ways to tune the parameters. If you meant this with "manuals", then you're already aware of them. The instructions suggest tuning the chopper parameters individually (as in spreadCycle separately from stealthChop), likely in order to reduce the number of variables.

I recently spent couple of weeks tuning the parameters and I found an oscilloscope with a current probe to be a vital help.
What I saw was that at higher speeds to RMS current starts dropping and therefore also the applied torque drops.

Either way, maybe a couple of things you can try.
If possible. use a higher VMOT voltage. At higher speeds to back EMF starts to get significant and the available "effective voltage" may not be sufficient anymore to reach the configured coil current. It's still a bit weird however that an increased current setting would decrease performance.
It's possible that the chopper frequency may not be high enough to keep up with the higher velocities. I see your hysteresis is quite high but it's also entirely possible that these settings are what match your motor best. You'd have to test and preferable measure to see quantifiable differences. Off time affects the spreadCycle chopper frequency.
You may find some help with switching to classic chopper or fullstepping at a certain speed, but personally I didn't find fullstepping to help improve the performance. It actually did the opposite, but perhaps still worth testing.
I also noticed a specific resonance frequency that distorts the current wave. This was somewhat dependent on the current setting but it also depends on your motor characteristics.

If you want good torque at high speeds, it's also possible that this is hard to achieve with stepper motors. Or perhaps just your particular stepper motor model (high inductance vs low inductance and so forth).

I'm sorry I may not be able to provide anything concrete. There are quite a few variables involved and the optimization is done largely through testing with the intended setup and in the conditions that you expect to working in and it may even require test equipment most don't have access to.

mschorer, 2021-01-30:
Hey, thank you for such a detailed answer to my cloudy question! Especially happy that you were trying to understand what i want, not pointing out a) the obvious or b) mistakes in wording!!

- yes, i've read datasheets and app notes (summarized as "manuals" ;-))
- VMOT is 24V - aware of the back EMF issue. But i don't have a feel for the "mechanics" of it yet ... and also not much headroom to the 28(29?)V the tmc2209 supports
- isolating the factors is good scientific advise
Again, it comes down to doing methodical testing including all factors ..  :-(

Grrrr, and i was just trying to not again dive into second level issues ... cutting corners doesn't work ..