teemuatlut/TMCStepper #186: TMC5160 Wiring with arduino

Hammadiqbal12, 2021-03-16:
I have a TMC5160 V1.2 and I want to use it with arduino Nano, I tried the Simple example code but it didn't work for me. Please guide me in following parts.
1. Do I need to connect all the SPI pins or only CS, EN, DIR and STEP pins are enough for this example code? I'm using this
#define EN_PIN           6 // Enable
#define DIR_PIN          4 // Direction
#define STEP_PIN         5 // Step
#define CS_PIN           7 // Chip select

#define R_SENSE 0.075f 
TMC5160Stepper driver(CS_PIN, R_SENSE);

2. If I need to use SPI pins, please indicate the SPI pins on TMC5160.

Thank you.

teemuatlut, 2021-03-16:
Yes, you do need to connect SPI.
The SPI pins on the TMC5160 are SDI, SCK, CSN and SDO.

daniel-frenkel, 2021-04-03:
> please indicate the SPI pins on TMC5160

You need to read the datasheet of your TMC5160 to find the SDI, SCK, CSN and SDO pins