teemuatlut/TMCStepper #189: incompatible with your current board which runs on (esp8266)

junkmail90210, 2021-03-25:
Hi,

I am writing this as I am receiving errors while trying to compile your driver for the 2209. I admit I am way out of my depth. I currently have my program working with the accelstepper library. I am looking to add stall detection on my stepper (not 3d printer). I was reading here, https://github.com/teemuatlut/TMC2130Stepper/issues/1 that the interrupts have to be set for my architecture. What is that and how do I set it?

Also if I may ask, will the 2209 just throw up a high on the diag pin? or can I only access it from uart tx rx pins? Currently I can't get the diag to change when it stalls.

Thanks!

teemuatlut, 2021-03-25:
Typically it helps to include the error statement if you have one =)
For the stallguard polling to work properly you'll need to use your platforms timer interrupts to drive the step pin. Otherwise the motor will briefly stop for the duration of the communication and mess with the readings.
Another option instead of polling the value is to monitor the diag pin and only read the status information when the driver informs you to. You can read about the diagnostic outputs on chapter 15.6.
Stallguard does need to be configured through UART (TCOOLTHRS, TPWMTHRS, SGTHRS, maybe others).

junkmail90210, 2021-03-26:
I can't  find chapter 15.6, which document?
eek, that would help with the error! sorry about that! 
```Arduino: 1.8.5 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 256000, 4M (3M SPIFFS)"

WARNING: library TMCStepper claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (esp8266) architecture(s).
stallguard:49: error: expected constructor, destructor, or type conversion before '(' token

 ISR(TIMER1_COMPA_vect){

    ^

Z:\Documents\Arduino\stallguard\stallguard.ino: In function 'void setup()':

stallguard:60: error: 'driver' was not declared in this scope

   driver.beginSerial(115200);

   ^

stallguard:81: error: 'TCCR1A' was not declared in this scope

     TCCR1A = 0;// set entire TCCR1A register to 0

     ^

stallguard:82: error: 'TCCR1B' was not declared in this scope

     TCCR1B = 0;// same for TCCR1B

     ^

stallguard:83: error: 'TCNT1' was not declared in this scope

     TCNT1  = 0;//initialize counter value to 0

     ^

stallguard:84: error: 'OCR1A' was not declared in this scope

     OCR1A = 256;// = (16*10^6) / (1*1024) - 1 (must be <65536)

     ^

stallguard:86: error: 'WGM12' was not declared in this scope

     TCCR1B |= (1 << WGM12);

                     ^

stallguard:88: error: 'CS11' was not declared in this scope

     TCCR1B |= (1 << CS11);// | (1 << CS10);

                     ^

stallguard:90: error: 'TIMSK1' was not declared in this scope

     TIMSK1 |= (1 << OCIE1A);

     ^

stallguard:90: error: 'OCIE1A' was not declared in this scope

     TIMSK1 |= (1 << OCIE1A);

                     ^

Z:\Documents\Arduino\stallguard\stallguard.ino: In function 'void loop()':

stallguard:105: error: 'TIMSK1' was not declared in this scope

       if (read_byte == '0')      { TIMSK1 &= ~(1 << OCIE1A); digitalWrite( EN_PIN, HIGH ); }

                                    ^

stallguard:105: error: 'OCIE1A' was not declared in this scope

       if (read_byte == '0')      { TIMSK1 &= ~(1 << OCIE1A); digitalWrite( EN_PIN, HIGH ); }

                                                     ^

stallguard:106: error: 'TIMSK1' was not declared in this scope

       else if (read_byte == '1') { TIMSK1 |=  (1 << OCIE1A); digitalWrite( EN_PIN,  LOW ); }

                                    ^

stallguard:106: error: 'OCIE1A' was not declared in this scope

       else if (read_byte == '1') { TIMSK1 |=  (1 << OCIE1A); digitalWrite( EN_PIN,  LOW ); }

                                                     ^

stallguard:108: error: 'OCR1A' was not declared in this scope

     else if (read_byte == '+') { if (OCR1A > MAX_SPEED) OCR1A -= 20; }

                                      ^

stallguard:109: error: 'OCR1A' was not declared in this scope

     else if (read_byte == '-') { if (OCR1A < MIN_SPEED) OCR1A += 20; }

                                      ^

stallguard:116: error: 'driver' was not declared in this scope

     Serial.print(driver.SG_RESULT(), DEC);

                  ^

exit status 1
expected constructor, destructor, or type conversion before '(' token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```
Thanks for information, that makes it much clearer.

daniel-frenkel, 2021-04-03:
I have it running on ESP32, so it may work here. However, I use dual-core on the ESP32 as well. 

I created a course to teach you how to use this library, you can watch the free sections of it to learn how to use it properly

https://tinyurl.com/trinamic-course