teemuatlut/TMCStepper #191: TMC2209 sensorless homing issue

killerrasta89, 2021-04-19:
Hi all,

I try to write a homing function, that I could call when I want. 
The whole project will be a macro slider with a DSLR camera. 

Here is a simplified code from my project.

The "homing" function and the stall sense works for the first time, but after, it doesn't. The while loop doesn't start.
Can you help me why is it works just once, and how can I fix the code to call the homing function when I want?

Thanks in advance for your help.

```
#include <TMCStepper.h>

#define STALL_VALUE      150
#define TOFF_VALUE        5

#define STALL_PIN         3
#define DIR_PIN           6
#define STEP_PIN          7
#define EN_PIN            5
#define SW_RX             2
#define SW_TX             8
#define DRIVER_ADDRESS 0b00
#define R_SENSE 0.11f 

TMC2209Stepper driver(SW_RX, SW_TX, R_SENSE, DRIVER_ADDRESS);
bool shaft = false;

void setup()
{
  pinMode(EN_PIN, OUTPUT);
  pinMode(STEP_PIN, OUTPUT);
  pinMode(DIR_PIN, OUTPUT);
  pinMode(STALL_PIN, INPUT_PULLUP);
  digitalWrite(EN_PIN, LOW);

  Serial.begin(115200);
  driver.begin();
  driver.toff(TOFF_VALUE);
  driver.rms_current(600);
  driver.microsteps(16);
  driver.TCOOLTHRS(0xFFFFF);
  driver.SGTHRS(STALL_VALUE);
}

void loop() 
{
  homing();
  delay(4000);
}

void homing() 
{ 
  while(digitalRead(STALL_PIN)==LOW)
  {
    digitalWrite(STEP_PIN, HIGH);
    delayMicroseconds(100);
    digitalWrite(STEP_PIN, LOW);
    delayMicroseconds(100);
  }
    
  shaft = !shaft;
  driver.shaft(shaft);
  
  for (uint16_t i = 5000; i > 0; i--) 
  {
    digitalWrite(STEP_PIN, HIGH);
    delayMicroseconds(200);
    digitalWrite(STEP_PIN, LOW);
    delayMicroseconds(200);
  }
}
```

teemuatlut, 2021-04-19:
You left out what it does do?
Looking at your code, I don't see any reason why it wouldn't loop to another round, but I've seen the stallGuard measurement to "stick". I'd suggest getting the motor to speed before reading the diag pin status, as the sudden speed change can trigger the stall condition.
You can also print the pin status before entering the while loop to get more information.

killerrasta89, 2021-04-20:
Thank You very much.

The problem was that you metioned. The DIAG PIN was at HIGH level, when the while loop tryed to start. I put a short for loop before the while loop and it works like a charm in every loop.

```
for (uint16_t i = 100; i > 0; i--) 
  {
    digitalWrite(STEP_PIN, HIGH);
    delayMicroseconds(100);
    digitalWrite(STEP_PIN, LOW);
    delayMicroseconds(100);
  }
```

closed 2021-04-20 by killerrasta89