teemuatlut/TMCStepper #192: Confused as to how to wire ESP32 to TMC2208 using TMCStepper.

evanvlane, 2021-05-01:
I know this is kind of basic, but I'm trying to understand how to wire up the uC and stepper driver.

**tl;dr:** Does anyone have a basic wiring diagram for ESP32->TMC2208? I haven't been able to get it to wire up correctly yet, despite trying several combinations, and I'm worried I'm going to burn something out.

I'd like to wire my [DORHEA NodeMCU ESP32](http://medesign.seas.upenn.edu/index.php/Guides/ESP32-pins) to this [TMC2208](https://www.amazon.com/gp/product/B082LSQWZF), but the pinouts are a bit confusing. 

I'm trying to work off the Simple.ino sketch, and the linked pinout references. I don't see a hardware serial out for my ESP32 board, but I know ESP32 is supposed to have three hardware serials. Is it not broken out on my board, and if so should I go with software serial? Or is UART2 RX/TX the same as hardware serial? In that case, would I enable hardware serial, and change the hardware serial port to Serial2 like this?

```
...
#define SERIAL_PORT Serial2 // TMC2208/TMC2224 HardwareSerial port
#define DRIVER_ADDRESS 0b00 // TMC2209 Driver address according to MS1 and MS2

#define R_SENSE 0.11f // Match to your driver
                      // SilentStepStick series use 0.11
                      // UltiMachine Einsy and Archim2 boards use 0.2
                      // Panucatt BSD2660 uses 0.1
                      // Watterott TMC5160 uses 0.075

// Select your stepper driver type
//TMC2130Stepper driver(CS_PIN, R_SENSE);                           // Hardware SPI
//TMC2130Stepper driver(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK); // Software SPI
//TMC2660Stepper driver(CS_PIN, R_SENSE);                           // Hardware SPI
//TMC2660Stepper driver(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK);
//TMC5160Stepper driver(CS_PIN, R_SENSE);
//TMC5160Stepper driver(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK);

TMC2208Stepper driver(&SERIAL_PORT, R_SENSE);                     // Hardware Serial
//TMC2208Stepper driver(SW_RX, SW_TX, R_SENSE);                     // Software serial
//TMC2209Stepper driver(&SERIAL_PORT, R_SENSE, DRIVER_ADDRESS);
//TMC2209Stepper driver(SW_RX, SW_TX, R_SENSE, DRIVER_ADDRESS);
...
```

Is the software slave clock (`SW_SCK`) the same thing as CLK on the TMC2208? And what is PDN? It's not mentioned in any of this stuff, but I found it named as "power down control" in some online tutorials, but it's not used nor explained.

teemuatlut, 2021-05-01:
https://circuits4you.com/2018/12/31/esp32-hardware-serial2-example/
> Like all peripherals, the pins for the UARTs can be logically mapped to any of the available pins on the ESP32

From the TMC library's point of the view the ESP32 platform does not support SW serial because there is no need for the SoftwareSerial class but instead you can define a serial class and give it the pins you want to use.

---

`SW_SCK` is for software SPI.
`PDN` is for controlling the automatic standstill current reduction but it also doubles as UART IO.

evanvlane, 2021-05-03:
Thanks for taking the time to respond!

I think I've made some progress, but I'm still not getting the TMC2208 driver to communicate or turn the motor.

Here's my code:
```
#include <TMCStepper.h>
 
#define EN_PIN           35 // Enable
#define DIR_PIN          32 // Direction
#define STEP_PIN         34 // Step

#define SERIAL_PORT Serial2 // TMC2208/TMC2224 HardwareSerial port
//#define DRIVER_ADDRESS 0b00 // TMC2209 Driver address according to MS1 and MS2

#define R_SENSE 0.11f // Match to your driver
TMC2208Stepper driver(&SERIAL_PORT, R_SENSE);                     // Hardware Serial
void setup() {
  pinMode(EN_PIN, OUTPUT);
  pinMode(STEP_PIN, OUTPUT);
  pinMode(DIR_PIN, OUTPUT);
  digitalWrite(EN_PIN, LOW);      // Enable driver in hardware

                                  // Enable one according to your setup
//SPI.begin();                    // SPI drivers
Serial.begin(9600);
SERIAL_PORT.begin(115200,SERIAL_8N1,33,25);      // HW UART drivers
//driver.beginSerial(115200);     // SW UART drivers

  driver.begin();                 //  SPI: Init CS pins and possible SW SPI pins
                                  // UART: Init SW UART (if selected) with default 115200 baudrate
  driver.toff(5);                 // Enables driver in software
  driver.rms_current(600);        // Set motor RMS current
  driver.microsteps(16);          // Set microsteps to 1/16th

//driver.en_pwm_mode(true);       // Toggle stealthChop on TMC2130/2160/5130/5160
driver.en_spreadCycle(true);   // Toggle spreadCycle on TMC2208/2209/2224
//driver.pwm_autoscale(true);     // Needed for stealthChop
//Serial.write("Got here.");
Serial.write(driver.version());
}

bool shaft = false;

void loop() {
  // Run 5000 steps and switch direction in software
  for (uint16_t i = 5000; i>0; i--) {
    digitalWrite(STEP_PIN, HIGH);
    delayMicroseconds(160);
    digitalWrite(STEP_PIN, LOW);
    delayMicroseconds(160);
  }
  shaft = !shaft;
  driver.shaft(shaft);
}
```

And here is how I've wired it. I did bridge the solder jumper for PDN/UART on the bottom of the board, and wired up the RX/TX UART with a 1kohm resistor like the documentation says. 

V_MOT is a 20VDC power supply, and I'm using the 3.3VDC line from the ESP32 to supply VIO.
I'm not sure what I'm doing wrong. When I try to write out the driver.version() info, I just get `null`.

Do you have any pointers for what's wrong with my setup?

[image]

teemuatlut, 2021-05-03:
It's possible the ESP32 has some constraints that I haven't dealt with before and it took me a couple of hours to get a good response. Even now it's not completely stable.

https://github.com/teemuatlut/TMCStepper/blob/ddc0c121af9117c14c173427f1dae165d8409ea5/src/TMCStepper.h#L1017
I had to change `TMC2208Stepper::replyDelay = 0` in `TMCStepper.h`

Here's my tested code otherwise:
```cpp
#include <TMCStepper.h>

TMC2208Stepper driver(&Serial2, 0.11);

void setup() {
  Serial.begin(9600);
  while(!Serial);

  Serial2.begin(19200);
  driver.begin();
}

void loop() {
  Serial.println(driver.version(), HEX);
  delay(2000);
}
```

There's a 1kOhm SMD resistor at the back of the driver PCB.
[image]

evanvlane, 2021-05-03:
I was able to get it to communicate and get the motor turning!

It looks like this should work. What's unstable for you? The ESP32 successfully talking to the TMC2208?
Your fix seems to help a lot. I haven't had any drops in the limited time I've been able to test it, but if you're seeing something in particular I should watch out for please let me know. 

Thanks for the time you put into this. It's only partially a code issue and I really appreciate you taking the time to walk me through the non-code stuff. I donated a bit to your account to say thanks.

closed 2021-05-04 by evanvlane

henriquefchaves, 2023-04-26:
Could it be the same for ESP8266?

Amazing