teemuatlut/TMCStepper #193: Large integer for Ramp

miracuruzrh, 2021-05-14:
Hi

I have this code for generating a ramp which produces "large integer implicitly truncated to unsigned type [-Woverflow]
" Is there a way to have smaller numbers or to correct the error?

`    // RAMP parameters
    driver.A1(      500000);
    driver.V1(     8000000);
    driver.AMAX(    100000);
    driver.VMAX(   9000000);
    driver.DMAX(    700000);
    driver.D1(      100000);
    driver.VSTOP(       10);`

teemuatlut, 2021-05-14:
```cpp
    // RAMP parameters
    driver.A1(      500000ul); // This is uint16 so max is   65535
    driver.V1(     8000000ul); // This is uint20 so max is 1048575
    driver.AMAX(    100000ul); // This is uint16 so max is   65535
    driver.VMAX(   9000000ul);
    driver.DMAX(    700000ul); // This is uint16 so max is   65535
    driver.D1(      100000ul); // This is uint16 so max is   65535
    driver.VSTOP(       10ul);
```

miracuruzrh, 2021-05-14:
> ```c++
>     // RAMP parameters
>     driver.A1(      500000ul); // This is uint16 so max is   65535
>     driver.V1(     8000000ul); // This is uint20 so max is 1048575
>     driver.AMAX(    100000ul); // This is uint16 so max is   65535
>     driver.VMAX(   9000000ul);
>     driver.DMAX(    700000ul); // This is uint16 so max is   65535
>     driver.D1(      100000ul); // This is uint16 so max is   65535
>     driver.VSTOP(       10u );
> ```

Have Many thanks for your Help

closed 2021-05-14 by miracuruzrh