teemuatlut/TMCStepper #195: Issue with 5160 BOB v1.2

MahOraibi, 2021-05-25:
I tried most of the examples in the library including software SPI, TMC_AccelStepper and simple in the hope of getting the driver to move the motor but nothing worked.
my setup is the following:
- TMC 5160 BOB v1.2
- Teensy 4.1
- Nema 17 motor 

I'm getting the DRV_STATUS when using software SPI example but then nothing moves.

is there a special way to move the 5160 other than step (high and low) or using accelstepper to move it? 

I'm completely noob in programming these things and I wish that someone can help me with this. 

`/**
 * Author Teemu Mäntykallio
 * Initializes the library and runs the stepper motor.
 */

#include <TMCStepper.h>

#define EN_PIN           0 // Enable
#define DIR_PIN          2 // Direction
#define STEP_PIN         1 // Step
#define CS_PIN           20 // Chip select
#define SW_MOSI          11 // Software Master Out Slave In (MOSI)
#define SW_MISO          12 // Software Master In Slave Out (MISO)
#define SW_SCK           13 // Software Slave Clock (SCK)

#define SERIAL_PORT Serial1 // TMC2208/TMC2224 HardwareSerial port
#define DRIVER_ADDRESS 0b00 // TMC2209 Driver address according to MS1 and MS2

#define R_SENSE 0.075 // Match to your driver

// Select your stepper driver type

//TMC5160Stepper driver = TMC5160Stepper(CS_PIN, R_SENSE); //I've also tried this but didn't work either
TMC5160Stepper driver = TMC5160Stepper(CS_PIN,  SW_MOSI, SW_MISO, SW_SCK);

bool dir = true;

void setup() {
  SPI.begin();
	Serial.begin(9600);
	while(!Serial);
	Serial.println("Start...");
	driver.begin(); 			// Initiate pins and registeries
	driver.rms_current(600); 	// Set stepper current to 600mA. The command is the same as command TMC2130.setCurrent(600, 0.11, 0.5);
	driver.en_pwm_mode(1);  	// Enable extremely quiet stepping

	pinMode(EN_PIN, OUTPUT);
	pinMode(STEP_PIN, OUTPUT);
	digitalWrite(EN_PIN, LOW); 	// Enable driver in hardware

	Serial.print("DRV_STATUS=0b");
	Serial.println(driver.DRV_STATUS(), BIN);
}

void loop() {
	digitalWrite(STEP_PIN, HIGH);
	delayMicroseconds(10);
	digitalWrite(STEP_PIN, LOW);
	delayMicroseconds(10);
	uint32_t ms = millis();
	static uint32_t last_time = 0;
	if ((ms - last_time) > 2000) {
		if (dir) {
			Serial.println("Dir -> 0");
			driver.shaft(0);
		} else {
			Serial.println("Dir -> 1");
			driver.shaft(1);
		}
		dir = !dir;
		last_time = ms;
	}
}`

teemuatlut, 2021-06-02:
Can you try this code snippet here https://github.com/teemuatlut/TMCStepper/issues/157#issuecomment-721960004
You'll need to use another branch like the [TMC2300 branch](https://github.com/teemuatlut/TMCStepper/tree/tmc2300) or the [Release_v1 branch](https://github.com/teemuatlut/TMCStepper/tree/Release_v1) to compile, but with some tweaks can be made compatible with current master as well.

The main interesting points are that you have working communication and that your hardware pins are configured for step/dir control. `SD_MODE` and `SPI_MODE`.
The TMC5160 can be controlled with just SPI commands using the internal Motion Controller. You can read more about it in the datasheet.

MahOraibi, 2021-06-02:
Thank you very much for your help, I'll try it and get back to you. What's more interesting is I got it to work on arduino mega 2560 with SPI and it works as intended. the same exact code with changing the pins to reflect SPI pins on teensy 4.1 don't seem to work, or if it does in very strange way.

teemuatlut, 2021-06-02:
Teensy4.1 is a bit too fast for its own good and it is able to start sending the data much faster than the main target platforms. For now the Teensy4.x can be considered as experimental but will be properly supported after I get some other stuff out of the way first.
https://github.com/teemuatlut/TMCStepper/issues/179#issuecomment-787314689

MahOraibi, 2021-06-06:
I believe this is the problem, however I hope to get a solution around this in the near future because my project is quite big with lots of checks, cases, and functions. I don't think arduino will cope with all this. This is the reason behind selecting teensy 4.1. Honestly, there's no much use for this board until now. teensystepper library doesn't work with it, TMCStepper library doesn't work either. It's been a bad choice to select which was only for the ARM m7. With that being said, there's an option in tools>processor speed inside arduino to slow the teensy down, won't this work with the library? one more thing, when I run the tmc5160 code by itself it works perfectly fine, but when I use tmc2209 in normal mode (only as step and direction driver) and with fastAccelStepper library the tmc5160 starts acting weird, the motor starts to jitter and shake without proper movement. Have you encountered such instance before? this is on arduino mega by the way.

Thank you very much teemuatlut for your great help and support. You are really an amazing person.

teemuatlut, 2021-06-06:
```cpp
// Board: Teensy4.0

#include <Arduino.h>
#include <TMCStepper.h> // v0.7.1

constexpr uint8_t cs = 10;

TMC5160Stepper driver(cs, 0.075);

void setup() {
  Serial.begin(9600);
  SPI.begin();
  
  pinMode(cs, OUTPUT);
  digitalWrite(cs, HIGH);
}

void loop() {
  delay(1000);
  Serial.printf("Firmware version = 0x%2X\tEnabled = %u\tMode = %s\n",
    driver.version(),
    driver.drv_enn(),
    driver.sd_mode() ? "Step/Dir" : "Ramp Generator"
  );
}

void TMC2130Stepper::switchCSpin(bool state) {
  delayNanoseconds(100);
  digitalWrite(_pinCS, state);
  delayNanoseconds(100);
}
```

I don't quite follow your setup anymore now that you're also using the TMC2209. Are you saying you're using two types of drivers?

If you use the 5160 in Ramp generator mode then you don't need to use any accelstepper libraries.
If however you want to use the step/dir interface of the driver, then you also need to configure the hardware input on the driver for this purpose. Perhaps you've already done this.

If you're seeing jitter, make sure your step pulses are generated properly with enough high time and not at too high frequency.

MahOraibi, 2021-06-06:
Sorry I just noticed my description of the issue was rubbish.
basically what I'm trying to do is to control multiple steppers for multiple axes. I intend to use the following for the setup:
- tmc5160 for X axis as a motion controller setting (works perfectly in its own using arduino mega)
- tmc5160 for F axis using the motion controller setting as well( not tested as well).
- tmc2209 for Y axis using fastAccelStepper(a modified faster version of the original accelStepper library > works perfectly in its own)
- tmc2209 for Z axis using the aforementioned library. (not tested yet)

so currently the tmc5160 (x axis) works perfectly if it was the only board in the arduino, but when I introduced the tmc2209 for Y axis and fastAccelStepper library to control the movement of Y axis only, and while the 5160 is still on its previous settings. The 5160 started to behave differently with noticeable jitters and it didn't move to left or right. I assume fastAccelStepper is using interrupts or maybe port manipulation which may or may not be the cause of the issue but I cannot be sure simply because I don't have that good experience of the subject and because the library is written in a way a newbie like me can't understand.

teemuatlut, 2021-06-06:
I tested the UART communication to a 2209 to be working without issues on the Teensy as well so from the point of configuring the drivers with the library everything seems to be ok as long as that delay is added for the CS pin.

I can help you further if you're able to isolate the issue to a smaller test case I can use to replicate.

MahOraibi, 2021-06-08:
Thank you very much buddy. 2209 actually worked on teensy 4.0 without issues. Now I just need to apply your code for 5160 and see what happens. Last time I tried it I got some errors but I'll give it a try and experiment with it and keep you as my last resort when I run of all options :) I feel I asked you for too much already.

teemuatlut, 2021-06-09:
I uploaded your code and the motor started spinning with no issue.
Here's the hw setup in case you need to double check wiring.
[image]

MahOraibi, 2021-06-09:
I just tried it again using the above code and it worked fine with Teensy. I don't know what was the reason for not being able to move the motor before but it seems to work perfectly now. 
Huge thanks from the bottom of by heart for all your support Teemu. Without your help I couldn't have achieved this. You are the best.

closed 2021-06-09 by MahOraibi