teemuatlut/TMCStepper #197: TMC2660 Software Serial not working on Adafruit M4 (Samd51)

deladriere, 2021-05-29:
Hardware SPI works fine using :
`TMC2660Stepper driver = TMC2660Stepper(CS_PIN, R_SENSE);`
using : 
`TMC2660Stepper driver = TMC2660Stepper(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK);`
doesn't work
`Serial.println(driver.DRV_STATUS(), BIN);`
reports 11111111110011111111 when the main power is off
reports ....
10100001010000000000
10100001100000000000
10100001110000000000
10100010000000000000
10100010010000000000
10100010100000000000

when power is ON

What am I doing wrong?
Thanks!

here is my code 

```
#include <Arduino.h>
/**
 * Author Teemu Mäntykallio
 * Initializes the library and runs the stepper motor.
 */

#include <TMCStepper.h>
#include <AccelStepper.h>

#define EN_PIN 10   // Enable
#define DIR_PIN 11  // Direction
#define STEP_PIN 12 // Step
#define CS_PIN 13   // Chip select

#define SW_MOSI 1 // Software Master Out Slave In (MOSI)
#define SW_MISO 4 // Software Master In Slave Out (MISO)
#define SW_SCK 0  // Software Slave Clock (SCK)

#define R_SENSE 0.1f // Match to your driver                         \
                      // SilentStepStick series use 0.11              \
                      // UltiMachine Einsy and Archim2 boards use 0.2 \
                      // Panucatt BSD2660 uses 0.1                    \
                      // Watterott TMC5160 uses 0.075

// Select your stepper driver type

//TMC2660Stepper driver = TMC2660Stepper(CS_PIN, R_SENSE); // Hardware SPI
TMC2660Stepper driver = TMC2660Stepper(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK);

AccelStepper stepper = AccelStepper(stepper.DRIVER, STEP_PIN, DIR_PIN);

bool dir = true;

void setup()
{

  Serial.begin(9600);

  Serial.println("Start...");

  SPI.begin();

  driver.begin();          // Initiate pins and registeries
  driver.rms_current(600); // Set stepper current to 600mA. The command is the same as command TMC2130.setCurrent(600, 0.11, 0.5);
                           // driver.en_pwm_mode(1);   // Enable extremely quiet stepping
  driver.microsteps(16);

  pinMode(EN_PIN, OUTPUT);
  pinMode(DIR_PIN, OUTPUT);
  pinMode(STEP_PIN, OUTPUT);
  pinMode(CS_PIN, OUTPUT);

  pinMode(SW_MOSI, OUTPUT);
  pinMode(SW_MISO, INPUT_PULLUP);
  pinMode(SW_SCK, OUTPUT);

  digitalWrite(EN_PIN, LOW); // Enable driver in hardware

  Serial.print("DRV_STATUS=0b");
  Serial.println(driver.DRV_STATUS(), BIN);

  stepper.setMaxSpeed(30000);      // 100mm/s @ 80 steps/mm
  stepper.setAcceleration(500000); // 2000mm/s^2
  stepper.setEnablePin(EN_PIN);
  stepper.setPinsInverted(false, false, true);
  stepper.enableOutputs();
  stepper.moveTo(1000);
}

void loop()
{
  if (stepper.distanceToGo() == 0)
    stepper.moveTo(-stepper.currentPosition());

  stepper.run();
  Serial.println(driver.DRV_STATUS(), BIN);
  delay(30);
}
```

enrique-aaron, 2021-05-31:
Hi,

Your problem is that you are using Pins 0 and 1, for MOSI and for SCK.  And those pin are used for serial communication (Which you are using to print out the messages).

#define SW_MOSI 1 // Software Master Out Slave In (MOSI)
#define SW_SCK 0  // Software Slave Clock (SCK)

Cheers,

Enrique.

deladriere, 2021-06-09:
Thanks Enrique
On the SAMD21 Pins 0 & 1 are not used with the Serial 
I found the problem 
1) pins must be declared by :  (at first I thought your library would do it for us)
```
 pinMode(SW_MOSI, OUTPUT);
 pinMode(SW_MISO, INPUT_PULLUP);
 pinMode(SW_SCK, OUTPUT);
```

2) my mistake : declaring the pins after the driver.begin() doesn't work