teemuatlut/TMCStepper #199: TMC2130 Stall detection

jasin755, 2021-06-10:
Hello. I am using example code with 
```
    Serial.print("0 ");-
    Serial.print(driver.sg_result(), DEC);
    Serial.print(" ");
    Serial.println(driver.cs2rms(driver.cs_actual()), DEC);
```

I am getting something like `0 430 214`. Second number is decreasing when motor is breaking. It is ok. But for different speeds this number when motor is normal running is different. How can I correctly detect stall? When the second and third numbers exactly mean?
Thanks.

teemuatlut, 2021-06-12:
The third number is the actual momentary RMS current and in the example is used to demonstrate coolStep functionality where the motor current is automatically adjusted based on the axle load. The load is measured with stallGuard.
The second number is the raw stallGuard value read from the driver.

The stallGuard measured values depend on multiple things like speed, load, current and used motor. You will need to tune the stallGuard threshold value to these conditions.
You can detect stall either by polling the measured value and reacting based on the value, or you can configure the driver to pull a diagnostic pin high/low when the value falls to 0.

jasin755, 2021-06-13:
> The third number is the actual momentary RMS current and in the example is used to demonstrate coolStep functionality where the motor current is automatically adjusted based on the axle load. The [...]

Is here some example usage with diagnostic pin?

teemuatlut, 2021-06-13:
You can call `dial0_stall(true);` or `diag1_stall(true);` and the driver will assert the respective diagnostic pin when the readout falls to 0.

The pin can be a bit "sticky" and you just need to drive a few steps for the condition to clear.

jasin755, 2021-06-13:
I have connected diag from driver to D11 pin on Arduino. Code:
```
#define DIAG_PIN              11
#define STALL_VALUE      15

void setup() {
      pinMode(DIAG_PIN, INPUT);
      driver.begin();
      driver.toff(4);
      driver.blank_time(24);
      driver.rms_current(400); // mA
      driver.microsteps(8);
      driver.TCOOLTHRS(0xFFFFF); // 20bit max
      driver.diag1_stall(1);
      driver.sgt(STALL_VALUE);
}

void loop() {
    Serial.print("DIAG PIN: ");
    Serial.print(digitalRead(DIAG_PIN) ? "H" : "L");
    Serial.println("");
}
````
In log I see only: 
```
...
I see DIAG PIN: L
...
```

jasin755, 2021-06-13:
Ok. `driver.diag1_pushpull(1);` was missing. But is it possible send me or add to this repo. Some sample how handle diag? Because I often got stall value immediately :( It is not stable.