teemuatlut/TMCStepper #200: TMC22XX UART stepping example

Flyguy, 2021-06-11:
I trying to move step motor with the uart only.
I have used the folowed code:
```
TMC2209Stepper driver(&SERIAL_PORT, R_SENSE, DRIVER_ADDRESS);
void setup() {
  driver.begin();
  driver.toff(2); 
  driver.pdn_disable(true);
  driver.rms_current(600);
  driver.microsteps(16);
  driver.pwm_autoscale(true);
}
  void loop() {
   Serial.println("vactual");
   driver.VACTUAL(1000);
   delay(1000);
  }
```
With oscilloscope i have seen good datagram at tx. But driver did nothing....
What i doing wrong ?

teemuatlut, 2021-06-11:
Would be curious to see that you don't need to call `SERIAL_PORT.begin(19200);` as the `driver.begin()` does not do that for you. Perhaps you particular platform initializes it for you.

However, I don't see a call to pull the enable line low.

`driver.begin()` already sets `pdn_disable` so you don't need to do that yourself.

Flyguy, 2021-06-11:
Hello teemuatlut!
Thank you for your reply. 
I am sorry, I have dropped some info for you.
I have forgotten add it to this peace of code.
I have used   Serial1.begin(115200); Platform is esp8266. Driver is tmc2226
I trying to enable the driver in software. EN pin is always height with pull up 10k resistor.

teemuatlut, 2021-06-11:
The enable pin is active low, so you need to pull it gnd.

[image]

Flyguy, 2021-06-11:
[image]
But what about this notice ?

teemuatlut, 2021-06-11:
Sounds like a boot time condition you should be aware about but I've never seen it to be a problem if the EN pin would be pulled low when powering on.

Flyguy, 2021-06-11:
In other words you trying to say what i can't enable the driver in software ? (without hardware pins from mc)

teemuatlut, 2021-06-11:
One way or another you need to pull the enable line low in order to power on the coils.

MahOraibi, 2021-06-11:
Do you mind if I pop in for a question? are you trying to move the motor without other libraries like accelsteppr or step (high/low)? is that possible? I really would love to do that directly from the board rather than from another library.

Flyguy, 2021-06-12:
> 
> 
> Do you mind if I pop in for a question? are you trying to move the motor without other libraries like accelsteppr or step (high/low)? is that possible? I really would love to do that directly from the board rather than from another library.

I have tested it yesterday... It's so primitive. VACTUAL command will set a motor speed. Motor speed can't be big. Any acceleration will no work with it. Motor will miss all steps on 1-2 rev/sec.
In my current requirements big speed not needed. But I need count steps. I will don't use it.

thecockroach, 2023-09-15:
Hi @Flyguy did you can move by counting steps? Im in the same problem with a TMC2209. Tks

Flyguy, 2023-09-15:
I have not used it. And it was long time ago.. But I think you can try to
using the index pin for this.