teemuatlut/TMCStepper #201: Position Encoder implementation with TMC5160

MahOraibi, 2021-06-11:
Hello Teemuatlut, 

I don't want to flood the github with questions but I really can't find proper information about this and to me the datasheet is too lengthy to process and grasp, and possibly some functions are done differently in the library (easier way) than what the datasheet suggests.

An example of what I mean is position encoder implementation. I really have no clue how to do it so here are some questions I hope you can help me with:

1- I'd like to use an encoder to automatically feed the position the motor currently in and ideally the GoTo position for the motor to move to using very basic angle representation. Would that be possible? are there prerequisites or equations I must know of (an example of that is some libraries use 4096 value for 12bit encoders and divide it from the angle and so on)?

2- I currently have AS5048a link for datasheet is [HERE](https://ams.com/documents/20143/36005/AS5048_DS000298_4-00.pdf) and also planning to but MLX309 link for datasheet is [HERE](https://www.melexis.com/-/media/files/documents/datasheets/mlx90393-datasheet-melexis.pdf). I know these have A1, A2, and trigger pins. Are these the pins the 5160 need to interface with the encoder?

3- what would be a basic code or functions to get position, and move to the next position or pre-saved position?

Thank you very much in advance for your help

teemuatlut, 2021-06-12:
The library abstracts away the connection and the bit modification for the registers but it is not meant to replace the datasheet.

You can connect an encoder to a TMC5160 and use it with the ramp generator.
The encoder needs to output the phase lines and optionally also the index line. Neither of those encoder provides that so you'll need to connect those straight to the MCU. The "A" pins set the I2C address.
If you want to have the TMC chip handle the encoder interface, you'll need something like a [CUI AMT102-V](https://www.digikey.fi/product-detail/fi/cui-devices/AMT102-V/102-1307-ND/827015).

MahOraibi, 2021-06-13:
That's exactly right, but what I meant there are some things which are done easily in the library because it takes away the heavy coding from the user and turns it to a sample single line code. Therefore, the implementation with library will be quite easier than how the datasheet explains it. 

Thank you very much for your information and your suggestion, now I have a better understanding of the situation but I still have one last question. From my understanding the encoder you suggested is able to work under 2048ppi per quadrature, now if my understanding is correct this means it's only able to recognize 2048 steps of one revolution? meaning a motor with microstepping less than 1/16? or is this irrelevant and this motor can achieve accurate reading even with 128 or 256 microstepping? 

and if what I think is correct will [AS5047D ](https://ams.com/as5047d)be the better option?

teemuatlut, 2021-06-13:
2048 per quadrature means 8192 per one full rotation, or 0.044 degrees per pulse. You need to keep in mind that the position accuracy with microstepping is theoretical and the motors themselves have tolerances.

You can connect a AS5047D directly to the MCU for position feedback, but not to the TMC5160.

MahOraibi, 2021-06-13:
> You need to keep in mind that the position accuracy with microstepping is theoretical and the motors themselves have tolerances.

Could you please elaborate on this point because I didn't catch your meaning? You made my head explode on this because it seems a bad thing :)

teemuatlut, 2021-06-13:
The microsteps themselves may not be equidistant at the micron level. The motors themselves list "step accuracy +/-5%" ([example](https://www.omc-stepperonline.com/download/17HS13-0316S.pdf)). Useful torque is only applied when there is difference between stator and rotor. In other words, the rotor will try to follow the stator magnetic field and align itself with it and any force is only present if they do not align. The force depends on how big the difference is. This is all when talking about differences smaller than a fullstep. If the difference (due to load perhaps) is too great the motor will lose a step.
But since you're using an encoder your positioning accuracy should be as accurate as your feedback loop.

MahOraibi, 2021-06-18:
Thank you very much teemuatlut for your continuous help. Honestly, this is too technical for me to understand but from what I can tell this means if I used 2048 PPS encoder with a stepper motor that generates 51200 pulses per rotation I should have no issue at all in terms of reading accuracy or so right? I still will get the accurate position of the motor with that.

teemuatlut, 2021-06-18:
Yes, you should be just fine using an encoder like that.

MahOraibi, 2021-06-21:
Thank you very much Teemuatlut. This is really helpful and much appreciated.

closed 2021-06-21 by MahOraibi