teemuatlut/TMCStepper #209: Not compiling in Atom + PlatformIO

lordton, 2021-08-18:
In file included from .pio/libdeps/LPC1768/TMCStepper/src/source/SERIAL_SWITCH.cpp:1:
.pio/libdeps/LPC1768/TMCStepper/src/source/SERIAL_SWITCH.h:14:19: error: 'uint16_t' does not name a type
   14 |     SSwitch(const uint16_t pin1, const uint16_t pin2, const uint8_t address);
      |                   ^~~~~~~~
.pio/libdeps/LPC1768/TMCStepper/src/source/SERIAL_SWITCH.h:14:40: error: 'uint16_t' does not name a type
   14 |     SSwitch(const uint16_t pin1, const uint16_t pin2, const uint8_t address);
      |                                        ^~~~~~~~
.pio/libdeps/LPC1768/TMCStepper/src/source/SERIAL_SWITCH.h:14:61: error: 'uint8_t' does not name a type
   14 |     SSwitch(const uint16_t pin1, const uint16_t pin2, const uint8_t address);
      |                                                             ^~~~~~~
.pio/libdeps/LPC1768/TMCStepper/src/source/SERIAL_SWITCH.h:17:11: error: 'uint16_t' does not name a type
   17 |     const uint16_t p1;
      |           ^~~~~~~~
.pio/libdeps/LPC1768/TMCStepper/src/source/SERIAL_SWITCH.h:18:11: error: 'uint16_t' does not name a type
   18 |     const uint16_t p2;
      |           ^~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio/build/LPC1768/lib3a9/TMCStepper/source/SERIAL_SWITCH.cpp.o] Error 1
========================= [FAILED] Took 88.86 seconds =========================

teemuatlut, 2021-08-18:
Does this happen with v0.7.1 or 0.7.2?

ellensp, 2021-08-18:
A work around is to force the program to use 0.7.1 not 0.7.2 till this is fixed.
In Marlin edit ini/features.ini    HAS_TRINAMIC_CONFIG                    = TMCStepper@0.7.1

ellensp, 2021-08-18:
@teemuatlut only 0.7.2

teemuatlut, 2021-08-18:
What are the compile flags on LPC? The only obvious change would be where RasPi support was added and including Stream and SPI headers are conditional to Arduino and the version.

ellensp, 2021-08-18:
arm-none-eabi-g++ -o .pio/build/LPC1768/libcb0/TMCStepper/source/SERIAL_SWITCH.cpp.o -c -std=gnu++17 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -fno-common -fno-threadsafe-statics -Os -ffunction-sections -fdata-sections -fsingle-precision-constant -Wall -mthumb --specs=nano.specs --specs=nosys.specs -mcpu=cortex-m3 -g3 -fmax-errors=5 -DLPC176X_FRAMEWORK -DTARGET_LPC1768 -DF_CPU=100000000L -DMCU_LPC1768 -DPIO_PLATFORM_VERSION=1003UL -DPIO_FRAMEWORK_VERSION=2008UL -DPLATFORMIO=50101 -D__MARLIN_FIRMWARE__ -DNDEBUG -DU8G_HAL_LINKS -DBOARD_F_CPU=100000000L -DARDUINO_ARCH_LPC176X -DARDUINOLPC=10805 -I/home/user/.platformio/packages/framework-arduino-lpc176x/libraries/Servo/src -I.pio/libdeps/LPC1768/TMCStepper/src -IMarlin/src/HAL/LPC1768/include -IMarlin/src/HAL/LPC1768/u8g -I/home/user/.platformio/packages/framework-arduino-lpc176x/cores/arduino -I/home/user/.platformio/packages/framework-arduino-lpc176x/system/CMSIS/include -I/home/user/.platformio/packages/framework-arduino-lpc176x/system/CMSIS/lib -I/home/user/.platformio/packages/framework-arduino-lpc176x/system/lpc176x .pio/libdeps/LPC1768/TMCStepper/src/source/SERIAL_SWITCH.cpp

teemuatlut, 2021-08-18:
Thanks. It's defining `ARDUINOLPC=10805` and that's why the headers are not included.

teemuatlut, 2021-08-18:
https://github.com/teemuatlut/TMCStepper/releases/tag/v0.7.3