teemuatlut/TMCStepper #210: tmc5160 spi communication problem

unknownrobots, 2021-08-23:
Hi everyone, I bought two tmc5160 from bigtreetech more specifically these https://www.biqu.equipment/products/bigtreetech-tmc5160-v1-0-driver-spi-mode-silent-high-precision-stepstick-stepper-motor-driver-with-heatsink-for-skr-v1-3-gen-v1-4-reprap?variant=20225467711586 the problem is I connected the driver to an arduino uno, mega and nano and tried to use the tmc stepper library and tested the examples that come with the library, sometimes the motor rotated but when I tried to change some parameter like microstepping or current it didn't work, so I I think the problem is in the spi communication, in the example code I just changed the corresponding spi pins to my arduino and selected the driver model, in the driver connection part with the arduino I connected the driver spi pins to the pins from the arduino, the driver has two power inputs one called VM and one called VCC-IO, I was researching and basically VM would be the power for the motor and VCC-IO would be the 5v of the driver but some people say they don't need to turn on the VCC-IO because the VM input has a voltage regulator, I don't know if it's true sometimes it worked without 5v on the VCC-IO and sometimes not, I don't know what's happening I've never used this driver and there's little information on how to use it without a 3d printer motherboard. 

```
This the example code I tried to use.
/**
 * Author Teemu Mäntykallio
 * Initializes the library and runs the stepper
 * motor in alternating directions.
 */

#include <TMCStepper.h>

#define EN_PIN           41 // Enable
#define DIR_PIN          3 // Direction
#define STEP_PIN         4 // Step
#define CS_PIN           53 // Chip select
#define SW_MOSI          51 // Software Master Out Slave In (MOSI)
#define SW_MISO          50 // Software Master In Slave Out (MISO)
#define SW_SCK           52 // Software Slave Clock (SCK)
#define SW_RX            63 // TMC2208/TMC2224 SoftwareSerial receive pin
#define SW_TX            40 // TMC2208/TMC2224 SoftwareSerial transmit pin
#define SERIAL_PORT Serial1 // TMC2208/TMC2224 HardwareSerial port
#define DRIVER_ADDRESS 0b00 // TMC2209 Driver address according to MS1 and MS2

#define R_SENSE 0.11f // Match to your driver
                      // SilentStepStick series use 0.11
                      // UltiMachine Einsy and Archim2 boards use 0.2
                      // Panucatt BSD2660 uses 0.1
                      // Watterott TMC5160 uses 0.075

// Select your stepper driver type
//TMC2130Stepper driver(CS_PIN, R_SENSE);                           // Hardware SPI
//TMC2130Stepper driver(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK); // Software SPI
//TMC2660Stepper driver(CS_PIN, R_SENSE);                           // Hardware SPI
//TMC2660Stepper driver(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK);
//TMC5160Stepper driver(CS_PIN, R_SENSE);
TMC5160Stepper driver(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK);

//TMC2208Stepper driver(&SERIAL_PORT, R_SENSE);                     // Hardware Serial
//TMC2208Stepper driver(SW_RX, SW_TX, R_SENSE);                     // Software serial
//TMC2209Stepper driver(&SERIAL_PORT, R_SENSE, DRIVER_ADDRESS);
//TMC2209Stepper driver(SW_RX, SW_TX, R_SENSE, DRIVER_ADDRESS);

void setup() {
  pinMode(EN_PIN, OUTPUT);
  pinMode(STEP_PIN, OUTPUT);
  pinMode(DIR_PIN, OUTPUT);
  digitalWrite(EN_PIN, LOW);      // Enable driver in hardware

                                  // Enable one according to your setup
  SPI.begin();                    // SPI drivers
//SERIAL_PORT.begin(115200);      // HW UART drivers
//driver.beginSerial(115200);     // SW UART drivers

  driver.begin();                 //  SPI: Init CS pins and possible SW SPI pins
                                  // UART: Init SW UART (if selected) with default 115200 baudrate
  driver.toff(5);                 // Enables driver in software
  driver.rms_current(600);        // Set motor RMS current
  driver.microsteps(16);          // Set microsteps to 1/16th

//driver.en_pwm_mode(true);       // Toggle stealthChop on TMC2130/2160/5130/5160
//driver.en_spreadCycle(false);   // Toggle spreadCycle on TMC2208/2209/2224
  driver.pwm_autoscale(true);     // Needed for stealthChop
}

bool shaft = false;

void loop() {
  // Run 5000 steps and switch direction in software
  for (uint16_t i = 5000; i>0; i--) {
    digitalWrite(STEP_PIN, HIGH);
    delayMicroseconds(160);
    digitalWrite(STEP_PIN, LOW);
    delayMicroseconds(160);
  }
  shaft = !shaft;
  driver.shaft(shaft);
}
```

gedeon93, 2021-09-22:
Upon revisiting this hub, I saw you created this issue just days after I created mine: https://github.com/teemuatlut/TMCStepper/issues/208#issue-972292934

I'm thinking the problem with the BigTreeTech 5160 is indeed SPI related. Interesting that you tested the board without `VCC_IO` as I never thought to try that. 

Unfortunately, there does not seem to be much information on the TMC5160s that helps hobbyists like us to fully understand how to get things working. I am too far into developing my system's first prototype that I am needing to stick with the Arduino boards until this current prototype development cycle of mine is complete. Otherwise, I would be testing an entirely different suite of microcontrollers to pinpoint whether or not the issue is with ATMega.

unknownrobots, 2021-09-23:
When i was looking if anyone had the same problem i found your thread, i agree with you have little documentation on how to use the drivers without a dedicated board, i changed some values ​​and my motor moves but i don't know it's too weird it just seems to be working like a normal driver i can't activate stealth chop or stall guard, lately i tried using an esp8266 because once i had problems with the spi of an arduino of mine so i tried to use but same arduino problems if i find any solution I'll tell you :)