teemuatlut/TMCStepper #212: 4 nema 17 steppers in parallel on 1 tmc5160

coolerooney, 2021-09-15:
Hi

My special application needs 4 small nema 17 motors driving 1 central axle 
Driver is tmc 5160 3Amps, 24 V

movement is troublesome
I am a bit stumped, 70 watts should be more yhan enough
Suggestions, settings?

thx!

Best

v

teemuatlut, 2021-09-22:
I don't think there are any special settings that would help you.
You can increase the current if you're not yet overheating or if you have the space use longer motors with more torque.
After that it's bigger motors or find a way to have an additional driver.
And of course make sure you have no mechanical losses anywhere.

coolerooney, 2021-09-22:
Hi Teemuatlut

We “solved” it, we now have 4 nema17 steppers in series, torque is fine, so is temp,
But the pps do not correspond anymore… 51200 is not 1rpm - any ideas
The project is a record player, 33 rpm, about 28k steps, I have the Trinamic TMCL IDE installed, (with 5160 eval kit) with a motion calculator, they specify about 29k steps, but not as pps but ppt- supposedly this has something to do with the mHz crystals
I practice we need a setting about 10k steps higher, to the to the 33 rpm

How/where can we control the stealthchop feature in your software?

Thx in advance

Best,

Carl

teemuatlut, 2021-09-22:
The chapter 12.1 gives a more detailed explanation of the conversions to real world unit but to explain how I've understood it is that the driver doesn't know anything about time as in seconds or milliseconds but what it does know is clock cycles. Unfortunately I can't tell you how the `t` is defined in the VMAX definition but it is somehow derived from f_clk.
With the help from the datasheet I too get to a value around VMAX=40000 for 33rpm.
[image]
The f_clk may not in reality be exactly 12MHz so if it matters you may want to use a more specific calibrated value.

StealthChop can be enabled with `.en_pwm_mode(true)` and if you want to change the parameters, all of them are exposed from the pwmconf register.