teemuatlut/TMCStepper #220: undefined refrence when included as a component with esp-idf

W-M-D, 2021-10-30:
I'm not quite sure what's going on here. I'm using release and i can't compile with esp-idf.

`/home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC2209Stepper.cpp:20: undefined reference to `TMC_HAL::TMC_UART::begin(unsigned int)'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/TMCStepper/libTMCStepper.a(TMC_UART.cpp.obj):(.literal._ZNK7TMC_HAL8TMC_UART18WaitForInhibitTimeEv+0x0): undefined reference to `TMC_HAL::TMC_UART::getTime() const'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/TMCStepper/libTMCStepper.a(TMC_UART.cpp.obj):(.literal._ZN7TMC_HAL8TMC_UART5writeEhj+0x4): undefined reference to `TMC_HAL::TMC_UART::preWriteCommunication()'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/TMCStepper/libTMCStepper.a(TMC_UART.cpp.obj):(.literal._ZN7TMC_HAL8TMC_UART5writeEhj+0x8): undefined reference to `TMC_HAL::TMC_UART::serial_write(void const*, signed char)'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/TMCStepper/libTMCStepper.a(TMC_UART.cpp.obj):(.literal._ZN7TMC_HAL8TMC_UART5writeEhj+0xc): undefined reference to `TMC_HAL::TMC_UART::postWriteCommunication()'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/TMCStepper/libTMCStepper.a(TMC_UART.cpp.obj):(.literal._ZN7TMC_HAL8TMC_UART15sendReadRequestERNS0_11ReadRequestE+0x8): undefined reference to `TMC_HAL::TMC_UART::available()'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/TMCStepper/libTMCStepper.a(TMC_UART.cpp.obj):(.literal._ZN7TMC_HAL8TMC_UART15sendReadRequestERNS0_11ReadRequestE+0xc): undefined reference to `TMC_HAL::TMC_UART::serial_read(void*, signed char)'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/TMCStepper/libTMCStepper.a(TMC_UART.cpp.obj):(.literal._ZN7TMC_HAL8TMC_UART4readEh+0x0): undefined reference to `TMC_HAL::TMC_UART::preReadCommunication()'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/TMCStepper/libTMCStepper.a(TMC_UART.cpp.obj):(.literal._ZN7TMC_HAL8TMC_UART4readEh+0x4): undefined reference to `TMC_HAL::TMC_UART::postReadCommunication()'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/TMCStepper/libTMCStepper.a(TMC_UART.cpp.obj): in function `TMC_HAL::TMC_UART::WaitForInhibitTime() const':
/home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:38: undefined reference to `TMC_HAL::TMC_UART::getTime() const'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/TMCStepper/libTMCStepper.a(TMC_UART.cpp.obj): in function `TMC_HAL::TMC_UART::write(unsigned char, unsigned int)':
/home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:50: undefined reference to `TMC_HAL::TMC_UART::preWriteCommunication()'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:53: undefined reference to `TMC_HAL::TMC_UART::serial_write(void const*, signed char)'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:53: undefined reference to `TMC_HAL::TMC_UART::postWriteCommunication()'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:54: undefined reference to `TMC_HAL::TMC_UART::getTime() const'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/TMCStepper/libTMCStepper.a(TMC_UART.cpp.obj): in function `TMC_HAL::TMC_UART::sendReadRequest(TMC_HAL::TMC_UART::ReadRequest&)':
/home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:60: undefined reference to `TMC_HAL::TMC_UART::getTime() const'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:67: undefined reference to `TMC_HAL::TMC_UART::serial_write(void const*, signed char)'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:69: undefined reference to `TMC_HAL::TMC_UART::getTime() const'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:70: undefined reference to `TMC_HAL::TMC_UART::serial_write(void const*, signed char)'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/TMCStepper/libTMCStepper.a(TMC_UART.cpp.obj): in function `TMC_HAL::TMC_UART::WriteDatagram::WriteDatagram()':
/home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/../source/TMC_UART.hpp:102: undefined reference to `TMC_HAL::TMC_UART::available()'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/TMCStepper/libTMCStepper.a(TMC_UART.cpp.obj): in function `TMC_HAL::TMC_UART::sendReadRequest(TMC_HAL::TMC_UART::ReadRequest&)':
/home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:82: undefined reference to `TMC_HAL::TMC_UART::serial_read(void*, signed char)'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:84: undefined reference to `TMC_HAL::TMC_UART::getTime() const'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:84: undefined reference to `TMC_HAL::TMC_UART::getTime() const'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:86: undefined reference to `TMC_HAL::TMC_UART::available()'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:91: undefined reference to `TMC_HAL::TMC_UART::serial_read(void*, signed char)'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:91: undefined reference to [... cut ...]
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:122: undefined reference to `TMC_HAL::TMC_UART::getTime() const'
/home/x/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/x/Projects/Company/x/Code/ESP32/x-ESP32/components/TMCStepper/src/source/TMC_UART.cpp:124: undefined reference to `TMC_HAL::TMC_UART::getTime() const'
`

teemuatlut, 2021-10-31:
ESP IDF is only supported in the Release v1 branch and even there I can only say it compiles but isn't fully tested.

W-M-D, 2021-11-01:
ah, i'm using the release v1 so I'm not sure either :/ just compiling and initializing it.

teemuatlut, 2021-11-01:
It does seem like there is a missing definition for `TMC_HAL::TMC_UART::begin` but that only explains one of the errors. You should make sure the IDF_VER is defined. PlatformIO which I'm using for unit testing does define this but it's possible that the official IDE (?) does not.

For the `begin` method you can do this before it is fixed:
```diff
--- a/examples/TMCStepper_ESP_IDF/src/main.cpp
+++ b/examples/TMCStepper_ESP_IDF/src/main.cpp
@@ -37,7 +37,6 @@ void app_main() {
     enablePin.reset();

     begin_spi();
-    begin_serial();

     driver.begin();
     driver.rms_current(600);
@@ -70,9 +69,9 @@ void begin_spi() {
     ESP_ERROR_CHECK(spi_bus_add_device(SPI2_HOST, &dev_cfg, &SPI));
 }

-void begin_serial() {
+void TMC_HAL::TMC_UART::begin(uint32_t baudrate) {
     uart_config_t uart_config = {
-        .baud_rate = 19200,
+        .baud_rate = static_cast<int>(baudrate),
         .data_bits = UART_DATA_8_BITS,
         .parity    = UART_PARITY_DISABLE,
         .stop_bits = UART_STOP_BITS_1,

```

W-M-D, 2021-11-04:
actually seems to be an issue with the cmake file not including enough stuff for the idf build system to pick it up 
From bf9b2c0f44502179ac92b7e9b13772faaf794d13 Mon Sep 17 00:00:00 2001
From: matthew weger <matthew.d.weger@gmail.com>
Date: Thu, 4 Nov 2021 19:30:01 -0400
Subject: [PATCH 141/141] fixed cmakelists

---
 CMakeLists.txt | 8 +++++---
 1 file changed, 5 insertions(+), 3 deletions(-)

diff --git a/CMakeLists.txt b/CMakeLists.txt
index 0efbb93..354054f 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -1,4 +1,6 @@
-idf_component_register(SRC_DIRS src/source
-                       INCLUDE_DIRS "src"
-)
+idf_component_register(SRC_DIRS "src/source" "src/source/TMC_HAL"
+                       INCLUDE_DIRS "src" "src/source/"
+		       REQUIRES arduino-esp32
+		       )
+			
 target_compile_options(${COMPONENT_LIB} PRIVATE -Wfatal-errors)
-- 
2.33.1

W-M-D, 2021-11-04:
> It does seem like there is a missing definition for `TMC_HAL::TMC_UART::begin` but that only explains one of the errors. You should make sure the IDF_VER is defined. PlatformIO which I'm using for [...]

closed 2021-11-04 by W-M-D

W-M-D, 2021-11-04:
Ah it's still not compiling probably because i have both the arudino-esp32 included as a component. 

so esp and arudino-esp are fighting defines :facepalm:

W-M-D, 2021-11-05:
second fix lol 

'''diff
From 6eff6082734b774a2482c9f375a516310f343608 Mon Sep 17 00:00:00 2001
From: matthew weger <matthew.d.weger@gmail.com>
Date: Thu, 4 Nov 2021 20:07:26 -0400
Subject: [PATCH 142/142] ARDUINO component fix

---
 src/source/TMC_HAL/TMC_HAL_ESP_IDF.cpp | 2 ++
 1 file changed, 2 insertions(+)

diff --git a/src/source/TMC_HAL/TMC_HAL_ESP_IDF.cpp b/src/source/TMC_HAL/TMC_HAL_ESP_IDF.cpp
index 1364168..aaab5a4 100644
--- a/src/source/TMC_HAL/TMC_HAL_ESP_IDF.cpp
+++ b/src/source/TMC_HAL/TMC_HAL_ESP_IDF.cpp
@@ -1,5 +1,6 @@
 
 #if defined(IDF_VER)
+#if !defined(ARDUINO)
 
 #include <esp_log.h>
 #include "../../TMCStepper.h"
@@ -108,3 +109,4 @@ __attribute__((weak))
 void TMC_UART::postReadCommunication() {}
 
 #endif
+#endif
-- 
2.33.1
'''

W-M-D, 2021-11-05:
i have a feeling this is going to be really annoying going forward but we'll see lmao. I really want some arduino libs but am used to the esp-idf. GAH