teemuatlut/TMCStepper #225: ESP32 & TMC5160 BOB 1.2

rauber-hotzenplotz, 2021-12-12:
Hallo

Im trying to use the XTARGET and the motion controller to control my stepper motor.

teemuatlut posted an example for the exakt same problem.

https://github.com/teemuatlut/TMCStepper/issues/157#issuecomment-721960004

Unfortunatly it doesnt work for me. It might be a small problem, but my knowledge is limited...

i changed

teemuatlut, 2021-12-12:
Which branch are you using?
The example given uses the newer architecture that is not yet in the master branch.

rauber-hotzenplotz, 2021-12-12:
thank you for the quick reply .

I was using the master branch. 

I just installed the 'TMCStepper-Release_v1' and deleted 'TMCStepper-master'
which slightly changes the error messages...

> In function 'void setup()':
ESP32BOBSPI:23:3: error: 'IOIN_t' is not a member of 'TMC5160_n::IOIN_i<TMC5160Stepper>'
   TMC5160_n::IOIN_i<TMC5160Stepper>::IOIN_t ioin{driver.IOIN()};
   ^
ESP32BOBSPI:25:7: error: 'ioin' was not declared in this scope
   if (ioin.version == 0xFF || ioin.version == 0) {
       ^
ESP32BOBSPI:31:18: error: 'ioin' was not declared in this scope
   Serial.println(ioin.version);
                  ^
ESP32BOBSPI:46:10: error: 'class TMC5160Stepper' has no member named 'A1'
   driver.A1(1000);
          ^
ESP32BOBSPI:47:10: error: 'class TMC5160Stepper' has no member named 'V1'
   driver.V1(50000);
          ^
ESP32BOBSPI:51:10: error: 'class TMC5160Stepper' has no member named 'D1'
   driver.D1(1400);
          ^
ESP32BOBSPI:52:10: error: 'class TMC5160Stepper' has no member named 'VSTOP'
   driver.VSTOP(10);
          ^
exit status 1
'IOIN_t' is not a member of 'TMC5160_n::IOIN_i<TMC5160Stepper>'

rauber-hotzenplotz, 2021-12-13:
AWESOME !!!
Thank you very much.

It works ! It lives. You were really helping me a lot.

(i made a small donation.)

cbaugher, 2022-01-12:
> Which branch are you using? The example given uses the newer architecture that is not yet in the master branch.

Hi,
When will the new architecture make it into the master branch?  If not fairly soon, how stable is the new architecture now?

C|

teemuatlut, 2022-01-18:
There are a lot of changes and there doesn't seem to be major core breaking issues with the current release so I'm taking my time with it. I at least want to be sure that it works with the main platforms like AVR, LPC, STM and I'm slowly working towards building unit tests and CI for them. So essentially there's no timeline and it'll be released when it's good enough.

cbaugher, 2022-01-27:
> There are a lot of changes and there doesn't seem to be major core breaking issues with the current release so I'm taking my time with it. I at least want to be sure that it works with the main [...]

Ok, fair enough.  I only ask because I'm using it in a project that will have to be finished in the next couple months, and I would rather use an officially released version.  If the new one was going to be released in the next couple weeks I would switch to it, but sounds like not, so I'll stick with the current release.
Thanks
C|