teemuatlut/TMCStepper #226: help configuring the 2209 on coolstep thanks in advance

gauix4, 2021-12-21:
I've been trying to activate cool step on my printer for several days now, but I don't see any difference in torque when it is activated or not.
Here is my current configuration if you have any suggestions I am fully open thank you in advance.
TMC 2209  rsens 0.11

`#if HAS_DRIVER(TMC2209)
  template<char AXIS_LETTER, char DRIVER_ID, AxisEnum AXIS_ID>
  void tmc_init(TMCMarlin<TMC2209Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID> &st, const uint16_t mA, const uint16_t microsteps, const uint32_t hyb_thrs, const bool stealth, const chopper_timing_t &chop_init, const bool interpolate) {
  
  TMC2208_n::GCONF_t gconf{0}; 
  gconf.pdn_disable = true;      
  gconf.mstep_reg_select = true; 
  gconf.i_scale_analog = false;
  gconf.internal_rsense = false;
  //gconf.en_spreadcycle = !stealth;
  st.GCONF(gconf.sr);
  //st.stored.stealthChop_enabled = stealth;
    
  TMC2208_n::CHOPCONF_t chopconf{0}; //renitialiser les valeurs du registe 
  chopconf.tbl = 0b01; // blank_time = 24 voir doc
  chopconf.toff = chop_init.toff;
  chopconf.intpol = interpolate; 
     chopconf.hstrt = chop_init.hstrt - 1;
   #if ENABLED(SQUARE_WAVE_STEPPING)
      chopconf.dedge = true;
   #endif 
  st.CHOPCONF(chopconf.sr); 
  st.rms_current(mA, HOLD_MULTIPLIER); 
  st.microsteps(microsteps); 
  st.iholddelay(10); 
  st.TPOWERDOWN(128);
  st.SGTHRS(100); 
  st.blank_time(24);
  st.TCOOLTHRS(10); 

  TMC2209_n::COOLCONF_t coolconf{0};
  coolconf.semin = 15;   
  coolconf.semax = 15;  
  coolconf.seup = 1;     
  coolconf.sedn = 0b010; 
  coolconf.seimin = 1;
  st.COOLCONF(coolconf.sr); 

  TMC2208_n::PWMCONF_t pwmconf{0}; 
  pwmconf.pwm_lim = 12;
  pwmconf.pwm_reg = 8; 
  pwmconf.pwm_autograd = true;
  pwmconf.pwm_autoscale = true;
  pwmconf.pwm_freq = 0b00; 
  pwmconf.pwm_grad = 2;  
  pwmconf.pwm_ofs = 2; 
  st.PWMCONF(pwmconf.sr); 

     #if ENABLED(HYBRID_THRESHOLD)
      st.set_pwm_thrs(hyb_thrs);
    #else
      UNUSED(hyb_thrs);
    #endif

  st.GSTAT(0b111); // Clear
  delay(200);
}
#endif // TMC2209`

teemuatlut, 2021-12-22:
You may be able to tune spreadCycle to be more quiet but it won't match stealthChop. Or you can try the TMC2209 where stallGuard works with stealthChop.
But coolstep is not an easy feature you can turn on and expect it to work right away. You'll need to monitor the stallGuard and current values to tune each threshold and increments. The most likely outcome is that you either get skipped steps because it doesn't react fast enough or you get no benefit because it's maxed out most of the time.

gauix4, 2021-12-25:
in fact there is a bug in Marlin it does not work with STM32 processors
I tested with a lpc1768 it works perfectly

gauix4, 2021-12-25:
TMC2208_n::GCONF_t gconf{0};
    gconf.pdn_disable = true; // Use UART
    gconf.mstep_reg_select = true; // Select microsteps with UART
    gconf.i_scale_analog = false;
    gconf.en_spreadcycle = !stealth;
    st.GCONF(gconf.sr);
    st.stored.stealthChop_enabled = stealth;

    TMC2208_n::CHOPCONF_t chopconf{0};
    chopconf.tbl = 0b00; // blank_time = 24
    chopconf.toff = chop_init.toff;
    chopconf.intpol = interpolate;
    chopconf.hend = chop_init.hend + 3;
    chopconf.hstrt = chop_init.hstrt - 1;
    TERN_(SQUARE_WAVE_STEPPING, chopconf.dedge = true);
    st.CHOPCONF(chopconf.sr);

    st.rms_current(mA, HOLD_MULTIPLIER);
    st.microsteps(microsteps);
    st.iholddelay(10);
    st.TPOWERDOWN(128); // ~2s until driver lowers to hold current

    TMC2208_n::PWMCONF_t pwmconf{0};
    pwmconf.pwm_lim = 12;//1/12 jerk
    pwmconf.pwm_reg = 8;//1/15 valeur de modiffication empliture PWM
    pwmconf.pwm_autograd = true;
    pwmconf.pwm_autoscale = true;
    pwmconf.pwm_freq = 0b01;
    pwmconf.pwm_grad = 50;
    pwmconf.pwm_ofs = 50;
    st.PWMCONF(pwmconf.sr);

    COOLCONF_t coolconf{0};   //renitialiser les valeurs du registe
  coolconf.semin = 15;      // % de d'ogmentation du courent suivent l'effort 1 à 15  //__//
  coolconf.semax = 15;      // % de diminution du courent suivent l'effort  1 à 15  ////
  coolconf.seup =3;        // valeur d'intensiter de modification de courent automatique // 1,2,4,8//
  coolconf.sedn = 0b010;    // StallGuard4TM fonction , valeur d'incrementation de micro pas pour les changement de vitesse // 1/2/8/16/32 micro pas / voir doc
  coolconf.seimin = 1;      //courant minimum pour le contrôle intelligent du courant voir doc !  0 = desactiver autres valeurs disponible
  coolconf.sgt = - 0 ; //-64 à +63 : une valeur plus élevée rend StallGuard2 moins sensible et nécessite plus de couple pour indiquer un décrochage.
  coolconf.sfilt = 0; // 0 à 1 // 0 Mode standard, haute résolution temporelle pour StallGuard //1 Mode filtré, signal StallGuard2 mis à jour pour chaque quatre pas complets 
  st.COOLCONF(coolconf.sr); //envoyer le packet en UART

    //TERN(HYBRID_THRESHOLD, st.set_pwm_thrs(hyb_thrs), UNUSED(hyb_thrs));

    st.GSTAT(0b111); // Clear
    delay(200);

issuefiler, 2023-09-15:
> @teemuatlut: coolstep is not an easy feature you can turn on and expect it to work right away. You'll need to monitor the stallGuard and current values to tune each threshold and increments. The most likely outcome is that you either get skipped steps because it doesn't react fast enough or you get no benefit because it's maxed out most of the time.

 Thanks for confirming that coolStep™ is useless and should be disabled because it requires manual tuning and is prone to misconfiguration that causes poor motor torque and step losses, which is not worth risking to reduce the heat. I find your quick little advice very helpful, far more helpful than Trinamic’s cryptic datasheets and the wiki page of this repository.

closed 2023-09-16 by gauix4