teemuatlut/TMCStepper #229: 'class TMC5130Stepper' has no member named 'A1'

daniel-frenkel, 2022-01-05:
I am using the TMC5130 in positioning mode. When setting up these motor velocity parameters

```
  driver.RAMPMODE(0);
  driver.A1(100);
  driver.AMAX(300);
  driver.DMAX(800);
  driver.D1(3000);
  driver.VSTART(0);
  driver.VSTOP(10);
  driver.V1(2500);
  driver.VMAX(120000);  
```
I receive the following error:

```
error: 'class TMC5130Stepper' has no member named 'A1'
   driver.A1(100);
          ^
5130-test:38:10: error: 'class TMC5130Stepper' has no member named 'D1'
   driver.D1(3000);
          ^
5130-test:41:10: error: 'class TMC5130Stepper' has no member named 'V1'
   driver.V1(2500);
          ^
```

This is very strange because the class does contain all of these members, as seen [here](https://teemuatlut.github.io/TMCStepper/class_t_m_c5130_stepper.html). It's weird that only the values with a 1 are doing this. Any idea why?

teemuatlut, 2022-01-05:
The `A1` `V1` and `D1` were renamed to lower case because they clashed with namespace polluting macros from the Arduino framework. The doxygen files haven't been generated after the change.

daniel-frenkel, 2022-01-07:
> The `A1` `V1` and `D1` were renamed to lower case because they clashed with namespace polluting macros from the Arduino framework. The doxygen files haven't been generated after the change.

That fixed it. Thanks!

closed 2022-01-07 by daniel-frenkel