teemuatlut/TMCStepper #230: XTARGET for TMC5130 not writing

daniel-frenkel, 2022-01-07:
I have used this library with other drivers and everything works great. Right now I am trying to port my firmware that I wrote for my TMC5130 to this library and for some reason, the XTARGET is not writing.

I am using an ESP32 with a TMC5130 in positioning mode. Here's a simple sketch that for some reason is not sending a value to XTARGET. I know that the pins and the hardware work for sure because I have different firmware that works with this setup. But your library is much easier and better so I am trying to transition :)

```
#include <TMCStepper.h>

#define chipCS 27
#define SCLK 26
#define MISO_PIN 33
#define MOSI_PIN 25
#define ENABLE_PIN 32
#define R_SENSE 0.075f

TMC5130Stepper driver = TMC5130Stepper(chipCS, R_SENSE, MOSI_PIN, MISO_PIN, SCLK); // Software SPI

void setup() {

  Serial.begin(115200);
  delay(1000);
  pinMode(chipCS,OUTPUT);
  pinMode(ENABLE_PIN, OUTPUT);
  digitalWrite(chipCS,HIGH);
  digitalWrite(ENABLE_PIN,LOW);
  
  SPI.begin(SCLK,MISO_PIN,MOSI_PIN,chipCS); // Edit 'pins.h' to change pins
  
  driver.begin();
  driver.en_pwm_mode(0);
  driver.toff(5);
  driver.hstrt(5);
  driver.hend(3);
  driver.tbl(2);
  
  driver.RAMPMODE(0);
  driver.a1(100);
  driver.AMAX(300);
  driver.DMAX(800);
  driver.d1(3000);
  driver.VSTART(0);
  driver.VSTOP(10);
  driver.v1(2500);
  driver.VMAX(120000);

  driver.rms_current(2000);
  driver.XTARGET(100000);
  driver.XACTUAL(0);

  Serial.print("ENABLE_PIN: ");
  Serial.println(digitalRead(ENABLE_PIN));
  Serial.print("driver.RAMPMODE(): ");
  Serial.println(driver.RAMPMODE());
  Serial.print("driver.a1(): ");
  Serial.println(driver.a1());
  Serial.print("driver.AMAX(): ");
  Serial.println(driver.AMAX());
  Serial.print("driver.DMAX(): ");
  Serial.println(driver.DMAX());
  Serial.print("driver.d1(): ");
  Serial.println(driver.d1());
  Serial.print("driver.VSTART(): ");
  Serial.println(driver.VSTART());
  Serial.print("driver.VSTOP(): ");
  Serial.println(driver.VSTOP());
  Serial.print("driver.v1(): ");
  Serial.println(driver.v1());
  Serial.print("driver.VMAX(): ");
  Serial.println(driver.VMAX());
  Serial.print("driver.rms_current(): ");
  Serial.println(driver.rms_current());
  Serial.print("driver.XTARGET(): ");
  Serial.println(driver.XTARGET());
  Serial.print("driver.XACTUAL(): ");
  Serial.println(driver.XACTUAL());

}  
void loop()
{
driver.XTARGET(100000);
Serial.println(driver.XTARGET());
delay(5000);
driver.XTARGET(0);
Serial.println(driver.XTARGET());
delay(5000);
}

```

In the serial monitor, I am just getting this back

```
ENABLE_PIN: 1
driver.RAMPMODE(): 0
driver.a1(): 100
driver.AMAX(): 300
driver.DMAX(): 800
driver.d1(): 3000
driver.VSTART(): 0
driver.VSTOP(): 10
driver.v1(): 2500
driver.VMAX(): 120000
driver.rms_current(): 1965
driver.XTARGET(): 0
driver.XACTUAL(): 0
```

And in the loop, the value of XTARGET is always 0.

I've spent many hours trying to figure this out and just cannot. Maybe you have an idea?

teemuatlut, 2022-01-07:
You can try this as it has been reported as working. Even if it's originally for 5160 it will probably work for 5130 with only small modifications. You'll need to use the Release_v1 branch or make two edits for master branch.

https://github.com/teemuatlut/TMCStepper/issues/225#issuecomment-992633347

daniel-frenkel, 2022-01-07:
I replaced the library with the Release_v1 and I also tried 
```
TMC5160Stepper driver (chipCS, R_SENSE, MOSI_PIN, MISO_PIN, SCLK); // Software SPI
```
and
```
TMC5160Stepper driver (chipCS, R_SENSE, MOSI_PIN, MISO_PIN, SCLK); // Software SPI
```
Here is the error. It doesn't look like Release_  v1 have VSTART and VSTOP
```
5130-test:10:65: error: use of deleted function 'TMC5160Stepper::TMC5160Stepper(TMC_HAL::PinDef, TMC_HAL::PinDef, TMC_HAL::PinDef, TMC_HAL::PinDef, int8_t)'
 TMC5160Stepper driver (chipCS, R_SENSE, MOSI_PIN, MISO_PIN, SCLK); // Software SPI
                                                                 ^

   TMC5160Stepper(TMC_HAL::PinDef, TMC_HAL::PinDef, TMC_HAL::PinDef, TMC_HAL::PinDef, const int8_t link_index = -1) = delete;
   ^
5130-test.ino: In function 'void setup()':
5130-test:35:10: error: 'class TMC5160Stepper' has no member named 'VSTART'
   driver.VSTART(0);
          ^
5130-test:36:10: error: 'class TMC5160Stepper' has no member named 'VSTOP'
   driver.VSTOP(10);
          ^
5130-test:57:25: error: 'class TMC5160Stepper' has no member named 'VSTART'
   Serial.println(driver.VSTART());
                         ^
5130-test:59:25: error: 'class TMC5160Stepper' has no member named 'VSTOP'
   Serial.println(driver.VSTOP());
                         ^
```

teemuatlut, 2022-01-07:
The constructor has been updated in v1
https://github.com/teemuatlut/TMCStepper/blob/3205ba0024139d3ab83c533184bff5dc03591b4c/examples/Simple/Simple.ino#L27-L41