teemuatlut/TMCStepper #243: Comms issues between TMC5160-EVAL and Arduino Uno wifi rev2.

Vexo2, 2022-06-17:
Hi,
I'm trying to make a stepper motor move with an arduino Uno wifi rev2 board (ATmega4809) and a TMC5160-EVAL driver. I'm new to this. And this is the first time I use a driver. I power the arduino with 5V USB and the TMC5160-Eval with 24V. I found the based code from Teemualtut and adapted it to my pins. It's compile fine but the stepper motor does not move. I looked with an oscilloscope and there is no signal. So I deduced that the problem was with the communication (to confirm trying to read a register from the TMC5160-EVAL the value returned is 0 or 255. In the serial monitor I get :

[image]

I get it with the code below :

```
/*
 * Author Teemu Mäntykallio
 * Initializes the library and runs the stepper motor.
 *  Arduino Pins   Eval Board Pins
 * 11 MOSI            32 SPI1_SDI
 * 12 MISO            33 SPI1_SDO
 * 13 SCK               31 SPI1_SCK
 * 7 CS                   30 SPI1_CSN
 * 5 DIO                 8 DIO0 (DRV_ENN) Enable
 * GND                  23 CLK16
 * 9 DIO                17 REFL_DIR
 * 8 DIO                18 REFL_STEP
 * GND                 2 GND
 * +5V                +5V_USB
 */

#include <TMCStepper.h>

#define EN_PIN           5 // Enable
#define DIR_PIN          9 // Direction
#define STEP_PIN         8 // Step
#define CS_PIN           7 // Chip select
#define SW_MOSI          11 // Software Master Out Slave In (MOSI)
#define SW_MISO          12 // Software Master In Slave Out (MISO)
#define SW_SCK           13 // Software Slave Clock (SCK)

#define R_SENSE 0.075f // Match to your driver
                     // SilentStepStick series use 0.11
                     // UltiMachine Einsy and Archim2 boards use 0.2
                     // Panucatt BSD2660 uses 0.1
                     // Watterott TMC5160 uses 0.075

// Select your stepper driver type
//TMC2130Stepper driver = TMC2130Stepper(CS_PIN, R_SENSE); // Hardware SPI
//TMC2130Stepper driver = TMC2130Stepper(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK); // Software SPI
//TMC2660Stepper driver = TMC2660Stepper(CS_PIN, R_SENSE); // Hardware SPI
//TMC2660Stepper driver = TMC2660Stepper(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK);
//TMC5160Stepper driver = TMC5160Stepper(CS_PIN, R_SENSE);
TMC5160Stepper driver = TMC5160Stepper(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK);

bool dir = true;

void setup() {
  Serial.begin(9600);
  while(!Serial);
  Serial.println("Start...");
  driver.begin();       // Initiate pins and registeries
  driver.rms_current(600);  // Set stepper current to 600mA. The command is the same as command TMC2130.setCurrent(600, 0.11, 0.5);
  driver.en_pwm_mode(1);    // Enable extremely quiet stepping

  pinMode(EN_PIN, OUTPUT);
  pinMode(STEP_PIN, OUTPUT);
  digitalWrite(EN_PIN, LOW);  // Enable driver in hardware

  Serial.print("DRV_STATUS=0b");
  Serial.println(driver.DRV_STATUS(), BIN);
}

void loop() {
  digitalWrite(STEP_PIN, HIGH);
  delayMicroseconds(10);
  digitalWrite(STEP_PIN, LOW);
  delayMicroseconds(10);
  uint32_t ms = millis();
  static uint32_t last_time = 0;
  if ((ms - last_time) > 2000) {
    if (dir) {
      Serial.println("Dir -> 0");
      driver.shaft(0);
    } else {
      Serial.println("Dir -> 1");
      driver.shaft(1);
    }
    dir = !dir;
    last_time = ms;
  }
}
```
I've been stuck for a week, my mistake may be obvious but I can't see it. I am interested in any advice that can help me solve my problem.
This is the first time I've posted an issue on Github please excuse my syntax errors.

SobhanAbedi, 2022-06-17:
Hi. Assuming there are no other errors in your code, as far as I can see you have no acceleration built in to your movement (stepping) logic and you are trying to drive your stepper at very high rpm (20 µs of delay with 1/16 micro stepping on 200 s/r stepper gives you around 1000rpm). That's physically not possible. And you are also trying to do this in stealthchop mode (driver.en_pwm_mode(1);) which makes it even more impossible! For initial testing use much much slower movement (like 10rpm) or use stepping libraries like FastAccelStepper.

Vexo2, 2022-06-17:
Thank you very much for your answer I appreciate your help! 
I hadn't noticed that the speed given was too high and inconsistent with the stealtchop mode. Do you know at what speed the stealthchop can be used properly ? 
Thanks for your advice, I'll keep you posted on the progress after the test.

SobhanAbedi, 2022-06-18:
Honestly it depends on your stepper-driver-current combination but you can expect something like 120rpm on the higher side.

Vexo2, 2022-06-18:
All right, I understand how it works, thanks.
After implementing your advice the stepper still wouldn't turn. So I looked at the schematic to see the links between the components that are on the arduino wifi rev2 board. And I noticed that the pins for the SPI function for the arduino uno wifi rev2 were not placed on the same pins as for the arduino Uno rev3. Once the wiring was changed and your advice applied everything worked. In case someone need it i just put the equivalences between this 2 arduino boards.
For arduino Uno REV3 :
```
/*
 *  Arduino Pins   Eval Board Pins
 * 11 MOSI            32 SPI1_SDI
 * 12 MISO            33 SPI1_SDO
 * 13 SCK               31 SPI1_SCK
 * 7 CS                   30 SPI1_CSN
 * 5 DIO                 8 DIO0 (DRV_ENN) Enable
 * GND                  23 CLK16
 * 9 DIO                17 REFL_DIR
 * 8 DIO                18 REFL_STEP
 * GND                 2 GND
 * +5V                +5V_USB
 */
```

For Arduino Uno wifi REV 2 (On ICSP side)
```
/*
 *  Arduino Pins                    Eval Board Pins
 * 2 (SPI1 side) MOSI            32 SPI1_SDI
 * 1 (SPI1 side) MISO            33 SPI1_SDO
 * 3 (SPI1 side) SCK               31 SPI1_SCK
 * 13 CS                                 30 SPI1_CSN
 * 5 DIO                                 8 DIO0 (DRV_ENN) Enable
 * GND                                  23 CLK16
 * 9 DIO                                17 REFL_DIR
 * 8 DIO                                18 REFL_STEP
 * GND                                  20 SD_MODE
 * +5V                                   22 SPI_MODE   (SD_MODE=0 and SPI_MODE=1 enable SPI function)
 * GND                                  2 GND
 * +5V                                 +5V_USB
 */
```
[image]

closed 2022-06-19 by Vexo2