teemuatlut/TMCStepper #256: shaft() does not always work?

cranefist, 2022-10-24:
Iw been testing using UART to replace the DIR pin, but I have noticed this issue that the `shaft()` does not always seem to trigger.

Im keeping track of the `IFCNT()`, and it registers an UART command but the `shaft()` does not change.

Any ideas about this?

tomasz90, 2022-12-19:
Did you find the solution, I am having same problem, shaft is not changing direction.

daniel-frenkel, 2023-05-24:
Depending on which driver you are using, you can use the TMC2208/TMC2209 internal pulse generator and change the VACTUAL value from positive to negative to change direction. I think you should only set up the shaft direction once during setup

issuefiler, 2023-09-11:
You can’t replace the `DIR` input. Your `shaft` register is working normally. The problem is that your `DIR` (direction) pin is _floating_.