teemuatlut/TMCStepper #257: ESP32 SOFTWARE SERIAL USE

Rmdhn1, 2022-10-25:
It's possible to use up to 8 drivers with the ESP32 using 2 hardware serial and TMC2209 which can be set to differing UART addresses.
But because the esp32 only has 3 UARTs, one for usb, you can only connect two TMC2208s, is it possible to connect more using software serial?
https://github.com/plerup/espsoftwareserial

cranefist, 2022-10-31:
I noticed that in the datasheet there is explained a method on how to use more than 4 drivers in single UART port.
[image]
Page 18
https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V103.pdf

Rmdhn1, 2022-11-06:
That'd seems like a fun challenge to make it work with marlin