teemuatlut/TMCStepper #259: TMC2209 Stallguard example: only stopped motor gives usable serial monitor numbers

TrulyMarcus, 2022-10-30:
Using Bigtreetech tmc2209 driver with a ramps 1.4 board. The modifications I've made to TMC2209 Stallguard Example have been: uncommenting the ISR timer lines(except for the STEP_PORT ^= 1 line) and uncommenting TMC2209Stepper driver(SW_RX, SW_TX, R_SENSE, DRIVER_ADDRESS); I also changed the Serial.begin(250000);  to have (115200). 

The issue is the serial monitor only outputs 0 0 55 while the monitor is engaged (spinning). If I use my hand to stop the spinning motor the numbers never change from the 0 0 55.  When I stop it with the "0" entry it does give me data, it happens to repeat ~ 0 344 214 with my setup. The + - and 1 commands work as expected.

From what I've read from other Issues on here the 0 0 55 is not what you expect and is a sign of a lack of signal, but apparently I am getting something when the motor is stopped. Being a Ramps board there aren't many wires to check connections for. My spliced rx/tx wires with the 1k resistor are continuous (although the resistor reads at 980, not 1k. if it matters some how). I've changed motors, I've tried a different TMC2209. I've switched changed up the orientation of the RX tx wires, and tried the UART wire on the 5 pin instead of 4, even though from what I've read my driver isn't set up in that way.

Is 0 0 55 supposed to be the result from a spinning motor on serial monitor? If not, any ideas why I am not getting the real numbers instead? I figure the numbers should change if I stop the spindle by hand.

cranefist, 2022-10-31:
Check this thread, there are some working examples. And start by making sure you have a properly set UART connection and you can read the microsteps you set in it.
https://forum.arduino.cc/t/using-a-tmc2209-silent-stepper-motor-driver-with-an-arduino/666992/40

TrulyMarcus, 2022-11-03:
I messed around more with the original example code and commented out digitalWrite(STEP_PIN, !digitalRead(STEP_PIN)); and uncommented  STEP_PORT ^= 1 << STEP_BIT_POS; under the ISR time part and it works fine now. I still wish I knew _why_ that made the difference but it did. I found a different Stallguard example by Milan Divkovic, that uses VACTUAL instead of the timer to spin the motor, that also gave me the readout without hassle. Hope someone can find that useful.

I had seen that thread before and it was filled with a lot that didn't apply to me. Not that there couldn't be great related info in there but by creating a post I was hoping for some more direct guidance. I'm definitely still a beginner and wasn't sure how to use "making sure you have a properly set UART connection and you can read the microsteps you set in it." in a meaningful way. Suggestions like that can be hard to google for and may not even relate to exactly how the responder meant them. I'm glad to have gotten a reply but after scouring through so many threads for information I've gotten frustrated by half expressed ideas.

closed 2022-11-03 by TrulyMarcus