teemuatlut/TMCStepper #262: Live_tune.ino won't compile

aguaviva, 2023-01-29:
I tried to compile this sample in order to test the UART mode connection. It turns out it fails to compile and shows a number of issues:

- `STEP_PORT` is not defined, I'd be nice to have a comment saying that you need to set it to the port you are using. 
- `myStepper` is not defined, I renamed it to 'driver' but not sure

When I fix those I get lot's of errors

```
/home/raul/snap/arduino/current/Arduino/Live_tune/Live_tune.ino: In function 'void setup()':
Live_tune:145:12: error: 'class TMC2208Stepper' has no member named 'en_pwm_mode'
  myStepper.en_pwm_mode(true);
            ^~~~~~~~~~~
Live_tune:148:12: error: 'class TMC2208Stepper' has no member named 'pwm_ampl'; did you mean 'pwm_ofs'?
  myStepper.pwm_ampl(180);
            ^~~~~~~~
            pwm_ofs
Live_tune:153:15: error: 'class TMC2208Stepper' has no member named 'diag1_stall'
     myStepper.diag1_stall(1);
               ^~~~~~~~~~~
Live_tune:154:15: error: 'class TMC2208Stepper' has no member named 'diag1_pushpull'
     myStepper.diag1_pushpull(1);
               ^~~~~~~~~~~~~~
/home/raul/snap/arduino/current/Arduino/Live_tune/Live_tune.ino: In function 'void loop()':
Live_tune:248:14: error: 'class TMC2208Stepper' has no member named 'en_pwm_mode'
    myStepper.en_pwm_mode(arg);
              ^~~~~~~~~~~
Live_tune:252:14: error: 'class TMC2208Stepper' has no member named 'enc_commutation'
    myStepper.enc_commutation(arg);
              ^~~~~~~~~~~~~~~
```
It seems I am the only one coming across this? I am using
- Arduino IDE: 1.8.19
- SW stack: Arduino AVR boards: 1.8.6
- HW: Ramps 1.4 on a Atmega 2560

Many thanks for putting this lib together!! :)

daniel-frenkel, 2023-06-02:
diag1_stall is only on the TMC5160, not the TMC2208. Your code doesn't match up with the TMC2208

flabrosse, 2025-07-17:
Similar errors with a TMC5160 (and many others).  I replaced `driver` with `myStepper` but still have erros, the first one being about STEP_PORT, like the OP.