teemuatlut/TMCStepper #266: Arduino Uno + TMC2160 + Nema23 driver not working

MrBlackMagic, 2023-02-14:
Hello folks.

I am trying to get the "MKS TMC2160-OC V1.0" driver working with my Arduino Uno board & a Nema23 stepper.
I connected all wires from the driver SPI (CS, DO, DI and SCK) to the corresponding I/O ports on the arduino additionally to the EN, DIR and STEP pins. DIAG0 and DIAG1 are floating / not connected. Hardware DIP switches on the driver are all set to 0 (OFF).

When i run the code below, the stepper starts spinning counter clockwise but continously and not changing direction.
In general the stepper is not reacting to any changes i do to the driver in software. No matter what i set as microstepp or rms_current it stays the same. 

```
/**
 * Author Teemu Mäntykallio
 * Initializes the library and runs the stepper
 * motor in alternating directions.
 */

#include <TMCStepper.h>

#define EN_PIN           8 // Enable
#define DIR_PIN          2 // Direction
#define STEP_PIN         5 // Step

#define CS_PIN           13 // Chip select
#define SW_MOSI          3 // Software Master Out Slave In (MOSI)
#define SW_MISO          7 // Software Master In Slave Out (MISO)
#define SW_SCK           6 //Software Slave Clock (SCK)

#define R_SENSE 0.11f // Match to your driver
                      // SilentStepStick series use 0.11
                      // UltiMachine Einsy and Archim2 boards use 0.2
                      // Panucatt BSD2660 uses 0.1
                      // Watterott TMC5160 uses 0.075

// Select your stepper driver type

TMC2160Stepper driver = TMC2160Stepper(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK); //sofware SPI
//TMC2160Stepper driver = TMC2160Stepper(CS_PIN, R_SENSE);

void setup() {
  pinMode(EN_PIN, OUTPUT);
  pinMode(STEP_PIN, OUTPUT);
  pinMode(DIR_PIN, OUTPUT);
  digitalWrite(EN_PIN, LOW);      // Enable driver in hardware
                              
  SPI.begin();                    // SPI drivers
  Serial.begin(9600);

  driver.begin();                 //  SPI: Init CS pins and possible SW SPI pins
                                  // UART: Init SW UART (if selected) with default 115200 baudrate
  driver.toff(5);                 // Enables driver in software
  driver.rms_current(4600);        // Set motor RMS current
  driver.microsteps(32);          // Set microsteps to 1/16th

  driver.pwm_autoscale(true);     // Needed for stealthChop
  
  Serial.print("DRV_STATUS=0b");
  Serial.println(driver.DRV_STATUS(), BIN);
  Serial.print("version: " );
  Serial.println(driver.version());
  Serial.print("run status: " );
  Serial.println(driver.irun());
  Serial.print("run en stat: " );
  Serial.println(driver.encb_dcen_cfg4());
  Serial.print("toff: " );
  Serial.println(driver.toff());
}

bool dir = false;

void loop() {
  
  // Run 5000 steps and switch direction in software
  for (uint16_t i = 2000; i>0; i--) {
    digitalWrite(STEP_PIN, HIGH);
    delayMicroseconds(160);
    digitalWrite(STEP_PIN, LOW);
    delayMicroseconds(160);
  }
  Serial.println("DIRECTION");
  dir = !dir;
  driver.shaft(dir);
}

```

To get the stepper switching direction i have to add

```
 if (dir)
    digitalWrite(DIR_PIN,HIGH);  
   else
    digitalWrite(DIR_PIN,LOW);  
```

The serial output looks always like this (if i change toff in software it also stays 15):

```
DRV_STATUS=0b11111111111111111111111111111111
version: 255
run status: 31
run en stat: 1
toff: 15
DIRECTION
DIRECTION
DIRECTION
DIRECTION
DIRECTION
DIRECTION
```

I have no clue whats going on. Do i have to make additional hardware or software changes, am i missing something?

Many thanks ins advance!

SobhanAbedi, 2023-03-09:
https://github.com/makerbase-mks/MKS-Big-Current-Driver/issues/2#issuecomment-645096162
I have tested it and it works.
I really hope you have figured it out in past 3 weeks ;)

PS the direction thing has nothing to do with this library. There are two types of drivers. Some drivers expect pulled-down dir pin and others expect pulled-up. This Library is meant for software control of TMC chips and not for driving the motor. The tests are just to demonstrate that the steppers work. You need to use other libraries (like FastAccelStepper) to actually move the motor if you want acceleration control and highspeed operation.

MrBlackMagic, 2023-03-13:
Thanks for the hint and your briefly explanation!
We switched to ESP32 instead of Arduino and used ESP_FlexyStepper lib to successfully run the stepper as we need to. Additonally we are now using a closed loop board so that we do not longer require a separate module like the TMC.  
If i try to get the TMC to work i will defenitely try your idea with the resisitor!