teemuatlut/TMCStepper #272: Unable to get StallGuard to work

sidn007, 2023-05-22:
Hi, I want to setup StallGuard and CoolStep. So far this is the code. I'm unable to get any StallGuard result, it just displays 0.
Getting StallGuard and CoolStep up and running would be really helpful. Thanks.

#include <AccelStepper.h>
#include <MultiStepper.h>
AccelStepper stepper1(1, 11, 12); 
MultiStepper StepperControl; 
int position=0;
int Sleep = 13;
#include <TMCStepper.h>
TMC2209Stepper driver(&Serial1, 0.11f, 0b00);
using namespace TMC2209_n;
                                          
void setup() 
{
  pinMode(Sleep, OUTPUT);
  digitalWrite(Sleep, LOW);  
  Serial.begin(115200);
  Serial1.begin(115200);
  driver.begin();
  driver.toff(5);  
  driver.sedn(0b00);
  driver.rms_current(1000);
  driver.I_scale_analog(false);
  driver.en_spreadCycle(false);
  driver.intpol(true);
  driver.microsteps(0);
  driver.ihold(2);
  driver.irun(10);
  driver.iholddelay(5);
  driver.TCOOLTHRS(0xFFFFF);
  driver.semin(5);
  driver.semax(2);
  driver.sedn(0b01);
  driver.SGTHRS(60);
  delay(300);
}

void loop() 
{
  Serial.println(driver.SG_RESULT());
  delay(100);
}

daniel-frenkel, 2023-05-24:
StallGuard is quite complex and really needs to be understood before you can make it work. 

I wrote [this kindle book](https://amzn.to/3ostxHU) on the topic and cover StallGuard in great detail. I am currently working on a Udemy course on the topic but the book should teach you everything you need to know.

I wish I could help you really fast, but StallGuard is too complex to simply explain here.

daniel-frenkel, 2023-06-02:
And FYI, you can't use StallGuard AND CoolStep at the same time. It's one or the other